Ultimaker/Cura #718: Cannot install on **UBUNTU** 16.04 - slicing doesn't occur on pre-upgrade versions

gazhay, 2016-04-23:
Seeing as I wasn't clear and the last bug I reported was closed....

You cannot install any of the newer DEBs on *Ubuntu16.04* because cura depends on an older version than the default one shipped in *UBUNTU16.04*. 

Even forcing the install sees cura installed, but crashes on launch with python errors.

I had a previous version installed before the upgrade, and it has stopped slicing since the upgrade.
"Show slicer log..." does nothing, so I can only assume it is also a python related error.

thopiekar, 2016-04-23:
What is actually the problem? Do you get an error when trying to install the DEB found on Ultimaker's homepage?
Please make a screenshot of that error dialog or take a terminal and run ```sudo dpkg -i <path to DEB>```. This will often return a more useful error message than the most GUI tools.

Thanks

thopiekar, 2016-04-23:
Ah, now I remember. The problem is that there is no Python 3.4 on Ubuntu 16.04 anymore.
There are two different solutions:
1. Make a mixed installation using apt-pinning. There are instructions on the internet which give you instructions (often for Debian) how to do that. In our case we need to add Ubuntu Wily again with a low priority. So when you tell to install Cura it will install Python 3.4 from Wily because there is no Python 3.4 on Xenial. All the other packages won't be affected as APT will always try to take dependencies from Xenial as possible (and like we want here.)
2. Another alternative: Install Cura from my personal (unsupported by Ultimaker) PPA. For this go to https://launchpad.net/~thopiekar/+archive/ubuntu/cura and follow the instructions. All stable releases of Cura will land there. Currently I'm using another PPA built on the master codebase, so I won't notice if there are problems in the stable PPA. So just report problems here, but don't for get to say that you are using my PPA!
Oh, and don't forget to uninstall (purge) the old cura package by Ultimaker before using my PPA.

Feel free to give feedback about your experience here :wink:

Ghostkeeper, 2016-04-25:
We're having trouble upgrading to Python 3.5 due to the Windows build. It needs to compile Python plug-ins with a more modern version of Visual Studio that is not working properly with Qt.

gazhay, 2016-05-03:
thopiekar, thanks, I tried your repo but unfortunately I'm stuck using older versions of cura as I have a delta printer (Dreammaker overlord) and cannot configure it for cura 2.

Slicing did start happening again in 15.02 but hasn't persisted across reboots.

Thanks for your help

thopiekar, 2016-05-03:
Oh, I'm sorry about that. Maybe you can take your time and try to convert the settings for your printer which come with 15.02 to the latest cura (aka master). Just take a look at the available JSON files, make first changes and create a pull request.
Then you can discuss with the developers there about missing parameters, etc.

No problem and best regards :wink:

BerndJM, 2016-05-30:
Please excause this maybe stupid question, but I'm quite a greenhorn in using PPA's.
I've seen there was a update today but the version is still 2.1.0.... where the official beta is 2.1.1
Could you please enlighten me about this?

thopiekar, 2016-05-30:
I simply didn't increased the version number. But you can find the build date/time on the launchpad page. It is even appended to the version number, I guess. So everytime I update the code hosted at launchpad with GitHub (sadly have to do this manually) a build is triggered.

So in short: The packages say they are 2.1.0, but in truth they come with 2.1.1. 😉

BerndJM, 2016-05-30:
Thank you for clearing up.
So I can be sure that occuring bugs are "actual" bugs :-))

thopiekar, 2016-05-30:
Yes, I hope so :smile: The only difference between the .deb provided by Ultimaker and my PPA is that:
* All dependencies, like Qt5, etc. and Uranium, libArcus, etc. are provided in one *.deb as an software bundle. My PPA provides all dependencies separated into different packages. So e.g. the installed Qt5 libraries provided by Ubuntu are used and not another copy (BUT: the one provided in Ultimaker's .deb is tested and known to work well).
* Ultimaker's .deb comes with a Updater Plugin, which notifies you about new updates, but this plugin is removed in my PPA as APT should do this job very well (as long as I update the codebase on launchpad).

So in case of bugs, don't forget to mention that you are using my PPA :wink:

awhiemstra, 2016-05-31:
> So e.g. the installed Qt5 libraries provided by Ubuntu are used and not another copy (BUT: the one provided in Ultimaker's .deb is tested and known to work well).

One thing to note here is that there is a known issue with PyQt < 5.6 where destruction of QML items will trigger a deadlock in PyQt. This is one of the reasons why we ship our own version of Qt these days.

thopiekar, 2016-05-31:
@BerndJM I noticed that the splash screen still shows "master" on 2.1. I made a little addition to the packaging files, so the package version, here 2.1.2 will be shown instead.

The PR I added is not needed for me actually anymore, but the reason that "master" still shows up.

BerndJM, 2016-05-31:
@thopiekar 
would like to say thanks for your work, without that I've been not able to play along with the new cura on my actual Ubuntu system.
<joke_mode_on> it's a shame, with the 15.09.xx and 1.99 versions we poor Ubuntu users had an soooo much to "old" sytem with the 14.04LTS - and now with the 2.1.x versions we have a to "new" system with the 16.04LTS<joke_mode_off>

@awhiemstra 
... where destruction of QML items ...
could you please shortly explain (for a not python programmer) what this means, so I can better estimate if it's a bug when something strange happens?

sedwards2009, 2016-06-01:
@BerndJM 
I can answer your question about the QML related bug. To hit it you need to do something which changes the settings in right sidebar, such as switching to different machine. When you hit it the application will hang and become unresponsive. It is quite nasty.

MCWeid, 2016-06-14:
Hi @thopiekar,
I am new to Linux and tried to follow your instructions. I added your PPA, purged the cura package and installed it again using apt get commands. Unfortunately it is still not starting and gives the following error messages
<file:///usr/share/cura/resources/qml/Cura.qml:7 module "QtQuick.Layouts" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:4 module "QtQuick" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:8 module "QtQuick.Dialogs" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:6 module "QtQuick.Controls.Styles" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:5 module "QtQuick.Controls" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:7 module "QtQuick.Layouts" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:4 module "QtQuick" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:8 module "QtQuick.Dialogs" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:6 module "QtQuick.Controls.Styles" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:5 module "QtQuick.Controls" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:7 module "QtQuick.Layouts" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:4 module "QtQuick" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:8 module "QtQuick.Dialogs" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:6 module "QtQuick.Controls.Styles" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:5 module "QtQuick.Controls" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:7 module "QtQuick.Layouts" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:4 module "QtQuick" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:8 module "QtQuick.Dialogs" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:6 module "QtQuick.Controls.Styles" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:5 module "QtQuick.Controls" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:7 module "QtQuick.Layouts" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:4 module "QtQuick" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:8 module "QtQuick.Dialogs" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:6 module "QtQuick.Controls.Styles" is not installed
file:///usr/share/cura/resources/qml/Cura.qml:5 module "QtQuick.Controls" is not installed

QQmlComponent: Component is not ready
Traceback (most recent call last):
  File "/usr/bin/cura", line 30, in <module>
    app.run()
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 205, in run
    self.initializeEngine()
  File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 130, in initializeEngine
    self.engineCreatedSignal.emit()
  File "/usr/lib/python3/dist-packages/UM/Signal.py", line 102, in emit
    func(dest, *args, **kargs)
  File "/usr/lib/cura/plugins/ChangeLogPlugin/ChangeLog.py", line 90, in _onEngineCreated
    self.showChangelog()
  File "/usr/lib/cura/plugins/ChangeLogPlugin/ChangeLog.py", line 96, in showChangelog
    self._changelog_window.show()
AttributeError: 'NoneType' object has no attribute 'show'
>
What am I doing wrong?

Thank you for your help!

Kind regards Marc.

Ghostkeeper, 2016-06-15:
It seems that the Qt package in that PPA doesn't have the plug-ins we require.

thopiekar, 2016-06-15:
That's correct. Looks like I've missed an package that is already installed, when using KDE 5.

Try to: ```sudo apt install qml-module-qtquick-dialogs qml-module-qtquick-controls ```

@Ghostkeeper: I guess Uranium needs them most of all?

Ghostkeeper, 2016-06-15:
Both Uranium and Cura create those dialogues and controls.

MCWeid, 2016-06-16:
Hi @thopiekar,
this worked for me. Cura is now starting and slicing. Thank you for your help.

thopiekar, 2016-06-16:
No problem. Will add the missing dependencies to the packages later.
Thank you for your feedback 👍

hulkco, 2016-07-05:
Hi i Tried to add you PPA but my system dont work: 
W: El repositorio «http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial Release» no tiene un fichero de Publicación.
N: Los datos de un repositorio como este no se pueden autenticar y por tanto su uso es potencialmente peligroso.
N: Vea la página de manual apt-secure(8) para los detalles sobre la creación de repositorios y la configuración de usuarios.
E: Fallo al obtener http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu/dists/xenial/main/binary-amd64/Packages  404  Not Found
E: No se han podido descargar algunos archivos de índice, se han omitido, o se han utilizado unos antiguos en su lugar.
Can you helpme ?

Ghostkeeper, 2016-07-05:
An attempt to translate for someone who might be able to debug:

> Warning: The repository "http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial Release" doesn't have a publication date.
> The data from a such a repository couldn't be authenticated so it could be dangerous to use.
> See the manual page "apt-secure(8)" for details on creating and configuring repositories.
> Error: Couldn't obtain http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu/dists/xenial/main/binary-amd64/Packages 404 Not Found
> Error: Some index files couldn't be downloaded, were omitted, or have been deprecated.

hulkco, 2016-07-06:
I change the language to English, this the full error:
sudo apt-get update
[sudo] password for gsreynaga: 
Hit:1 http://ppa.launchpad.net/freecad-maintainers/freecad-stable/ubuntu xenial InRelease
Hit:2 http://mx.archive.ubuntu.com/ubuntu xenial InRelease                    
Ign:3 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial InRelease
Hit:4 http://mx.archive.ubuntu.com/ubuntu xenial-updates InRelease      
Hit:5 http://ppa.launchpad.net/js-reynaud/kicad-4/ubuntu xenial InRelease
Hit:6 http://mx.archive.ubuntu.com/ubuntu xenial-backports InRelease     
Hit:7 http://ppa.launchpad.net/teejee2008/ppa/ubuntu xenial InRelease
Hit:8 http://mx.archive.ubuntu.com/ubuntu xenial-security InRelease      
Hit:9 http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu xenial InRelease
Hit:10 http://ppa.launchpad.net/webupd8team/java/ubuntu xenial InRelease
Ign:11 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial Release
Ign:12 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 Packages
Ign:13 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main i386 Packages
Ign:14 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main all Packages
Ign:15 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en_US
Ign:16 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en
Ign:17 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-es
Ign:18 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:19 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main DEP-11 64x64 Icons
Ign:12 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 Packages
Ign:13 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main i386 Packages
Ign:14 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main all Packages
Ign:15 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en_US
Ign:16 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en
Ign:17 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-es
Ign:18 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:19 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main DEP-11 64x64 Icons
Ign:12 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 Packages
Ign:13 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main i386 Packages
Ign:14 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main all Packages
Ign:15 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en_US
Ign:16 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en
Ign:17 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-es
Ign:18 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:19 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main DEP-11 64x64 Icons
Ign:12 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 Packages
Ign:13 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main i386 Packages
Ign:14 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main all Packages
Ign:15 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en_US
Ign:16 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en
Ign:17 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-es
Ign:18 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:19 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main DEP-11 64x64 Icons
Ign:12 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 Packages
Ign:13 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main i386 Packages
Ign:14 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main all Packages
Ign:15 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en_US
Ign:16 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en
Ign:17 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-es
Ign:18 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:19 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main DEP-11 64x64 Icons
Err:12 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 Packages
  404  Not Found
Ign:13 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main i386 Packages
Ign:14 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main all Packages
Ign:15 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en_US
Ign:16 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-en
Ign:17 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main Translation-es
Ign:18 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main amd64 DEP-11 Metadata
Ign:19 http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial/main DEP-11 64x64 Icons
Reading package lists... Done
W: The repository 'http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu xenial Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://ppa.launchpad.net/gwibber-daily/ppa/ubuntu/dists/xenial/main/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.

Regards...

thopiekar, 2016-07-06:
Good morning,

you've got an old PPA enabled, which has been disabled. Just take 'sudo
software-properties-gtk' and remove the URL, which mentions "gwibber".

After that "apt-get update" should work again.

thopiekar, 2016-07-06:
Oh, and you can easily translate every application by adding the prefix "LANG=C" 😉
So this would result into "LANG=C sudo apt-get update"..

nickthetait, 2016-07-06:
That's a handy translation switch to know about :+1:

thopiekar, 2016-07-06:
I'm sure you could even ask me this while sleeping. Almost need it every time I have to read others console outputs :smile:

HyGy, 2016-09-01:
Will it work with the ubuntu 16.04 with the downloaded version from the original site? (I tried 2.3).

gazhay, 2016-09-01:
If its the same issue - some of us have tried pull requests to get the thing fixed, but none seem to get accepted.

For reference, the line I needed to change to make things work was documented here 
https://github.com/daid/LegacyCura/issues/1461

thopiekar, 2016-09-01:
The problem at the moment is not the same. Currently more recent Ubuntu versions come with Python 3.5 only, but our DEB needs 3.4 (nothing lower or higher).

In the future we'll use AppImages, which means everything will be bundled into one executable binary file. So no DEB's anymore. Just 'chmod a+x <our file>' and double click on it.
The only alternative to that is my personal PPA where I provide 'traditional' DEB packages.

I've been working on these AppImages and they are working now. It only needs a review next week and hopefully get releases on the next Beta release (if there are no further problems with it).

Regards

HyGy, 2016-09-01:
@thopiekar Can I test this bundle on my ubuntu 16.04?

thopiekar, 2016-09-17:
@HyGy: The usage of AppImage has been moved into future sprints. Can't decide on that, but you can try to use my PPA. It comes with 2.3 now.

Or are you using a non-Debian system?

HyGy, 2016-09-18:
Hi @thopiekar,

I added your personal ppa, and here is what I got:

Error:18 https://launchpad.net/~thopiekar/+archive/ubuntu/cura xenial/main amd64 Packages
  404  Not Found
Error:19 https://launchpad.net/~thopiekar/+archive/ubuntu/cura xenial/main i386 Packages
  404  Not Found

thopiekar, 2016-09-18:
@HyGy : There are instructions on the PPA page, see "Adding this PPA to your system".
In short you can add the PPA by:
* `sudo add-apt-repository ppa:thopiekar/cura`
or using the URL:
* `http://ppa.launchpad.net/thopiekar/cura/ubuntu`
in your sources.list.

HyGy, 2016-09-18:
Meanwhile I found it, thanks. :) I just wake up :) sorry.

thopiekar, 2016-09-18:
No problem :smile:
Have a nice day :wink:

HyGy, 2016-09-18:
Where can I set the diameter of the filament? In the materials I cant set anything.

@thopiekar it was just an afternoon sleeping :) Thanks anyway. :)))

HyGy, 2016-09-18:
Okay, I found it I need to duplicate one material and I can edit it, but I think all the other materials is not necessary for me, couse I'm using 1.75mm filament.

HyGy, 2016-09-18:
My first test print is running (okay, the second, first I forget put the autobedlevel g code to start gcode.

thopiekar, 2016-09-18:
Great! :+1:

HyGy, 2016-09-18:
The test print is success. The installation is good.

When I press open folder after slicing and saving to disk, it open 3 times the folder. :)

I printed an lm8uu bearing with a prusa i3:
- the outter surface is very good
- the first layer is a little bit bigger
- i want to pull it to a 8mm road, but it is very hard, and I can't move it on the road with my hand. (Sliced with slic3r I can move it).

I print a treefrog now.

HyGy

KlausHans, 2016-09-29:
Hello @thopiekar, 
i have troubles to get Cura to work on Ubuntu Gnome 16.04.
First i tried the .deb from the Ultimaker website and run in the Python 3.4 problem.
Then i tried your ppa. It starts, but i am not able to save settings, edit profiles etc. When i close it and reopen cura, then it asks me again wich printer i use. Starting it with sudo did not changed the behavior. Besides that it seems quite unstable, crashed two times in a short period of time. 
What am i missing? Thanks.

thopiekar, 2016-09-29:
Hey @KlausHans,

rule #1: never ever use 'sudo' if it is really needed.
If there are problems (same for other applications) take a terminal instead and run your application manually. Often there is a error message which mentions which problem is present.

Now back to your problem here: For me it sounds like you've messed up the directory/file permissions by using sudo (by using sudo on Cura or using sudo on another application). Please google how to repair your $HOME's permissions. You will find guides easily, I'm sure.
When that's done, I'll take another look on that for sure. (However it doesn't seem to be a software or packaging related problem so far.)

Regards

BerndJM, 2016-09-29:
For the rule #1 and the permissions @thopiekar  has right (i tried this unintented *gg*).
Running Cura as sudo and all the config folders belong to the user "root" so you're afterwards not able to save any configs as the normal user ...

thopiekar, 2016-09-29:
Thank you for confirmation.
I'm sure resetting all permissions of the whole $HOME directory is wise here, as @KlausHans probably already played with sudo in the past.

KlausHans, 2016-09-30:
Thanks, this was indeed the problem, never experienced it before. It works now.

The sudo-thing was just for testing after it did not work in the first place.

BerndJM, 2016-09-30:
For me the question came in mind:
why the hell writes Cura - started as root (via sudo) - the config files to "/home/user" and not to "/root"
???

thopiekar, 2016-09-30:
If I'm not wrong, running 'sudo' keeps your environment variables and they say that $HOME is at your home directory. So Cura will use your home directory at the end, too.
In comparison running 'sudo bash' and then cura will use root's home. But not sure whether bash adopts the old environment variables again..

sanderux, 2016-10-22:
@thopiekar any chance you could update your PPA to include the ultimaker 3?

probonopd, 2016-10-22:
@thopiekar any chance we'll eventually see the AppImages?

thopiekar, 2016-10-22:
@sanderux Yes, the sources for the UM3NetworkPrintingPlugin are now open. I was on vacation and now I'm back home. It should be available "soon".

@probonopd Can't say anything about it. Didn't take a look at the cura-build code for a while, but it is planned for 2.4 afaik.

thopiekar, 2016-10-23:
@sanderux : The UM3 plugin is now available:
```
cura-extra-plugin-um3networkprinting_2.3.0-0~201610231124~66~ubuntu16.04.1_all.deb
----------------------------------------------------------------------------------

 new debian package, version 2.0.
 size 29114 bytes: control archive=821 bytes.
     421 bytes,    11 lines      control              
     953 bytes,    10 lines      md5sums              
 Package: cura-extra-plugin-um3networkprinting
 Source: cura-extra-plugins
 Version: 2.3.0-0~201610231124~66~ubuntu16.04.1
 Architecture: all
 Maintainer: Thomas Karl Pietrowski <thopiekar@googlemail.com>
 Installed-Size: 134
 Depends: cura-plugin-api-3, python3-zeroconf
 Section: science
 Priority: optional
 Description: Cura plugin - UM3NetworkPrinting
  For more info see: https://github.com/Ultimaker/UM3NetworkPrintingPlugin

drwxr-xr-x root/root         0 2016-10-23 11:26 ./
drwxr-xr-x root/root         0 2016-10-23 11:26 ./usr/
drwxr-xr-x root/root         0 2016-10-23 11:26 ./usr/lib/
drwxr-xr-x root/root         0 2016-10-23 11:26 ./usr/lib/cura/
drwxr-xr-x root/root         0 2016-10-23 11:26 ./usr/lib/cura/plugins/
drwxr-xr-x root/root         0 2016-10-23 11:26 ./usr/lib/cura/plugins/UM3NetworkPrinting/
-rw-r--r-- root/root      6718 2016-10-23 11:24 ./usr/lib/cura/plugins/UM3NetworkPrinting/DiscoverUM3Action.py
-rw-r--r-- root/root     13412 2016-10-23 11:24 ./usr/lib/cura/plugins/UM3NetworkPrinting/DiscoverUM3Action.qml
-rw-r--r-- root/root     34520 2016-10-23 11:24 ./usr/lib/cura/plugins/UM3NetworkPrinting/LICENSE
-rw-r--r-- root/root     51080 2016-10-23 11:24 ./usr/lib/cura/plugins/UM3NetworkPrinting/NetworkPrinterOutputDevice.py
-rw-r--r-- root/root      9294 2016-10-23 11:24 ./usr/lib/cura/plugins/UM3NetworkPrinting/NetworkPrinterOutputDevicePlugin.py
-rw-r--r-- root/root       235 2016-10-23 11:24 ./usr/lib/cura/plugins/UM3NetworkPrinting/README.md
-rw-r--r-- root/root      4095 2016-10-23 11:24 ./usr/lib/cura/plugins/UM3NetworkPrinting/UM3InfoComponents.qml
-rw-r--r-- root/root       729 2016-10-23 11:24 ./usr/lib/cura/plugins/UM3NetworkPrinting/__init__.py
drwxr-xr-x root/root         0 2016-10-23 11:26 ./usr/share/
drwxr-xr-x root/root         0 2016-10-23 11:26 ./usr/share/doc/
drwxr-xr-x root/root         0 2016-10-23 11:26 ./usr/share/doc/cura-extra-plugin-um3networkprinting/
-rw-r--r-- root/root       233 2016-10-23 11:24 ./usr/share/doc/cura-extra-plugin-um3networkprinting/changelog.Debian.gz
-rw-r--r-- root/root      1701 2016-10-23 11:24 ./usr/share/doc/cura-extra-plugin-um3networkprinting/copyright

```

sanderux, 2016-10-24:
@thopiekar thanks for the update.
Not sure what changed but now cura wont run anymore:
```
uncaught exception has occurred!
Traceback (most recent call last):
  File "/usr/bin/cura", line 61, in <module>
    app = cura.CuraApplication.CuraApplication.getInstance()
  File "/usr/lib/python3/dist-packages/UM/Application.py", line 308, in getInstance
    Application._instance = cls()
  File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 158, in __init__
    UM.VersionUpgradeManager.VersionUpgradeManager.getInstance().setCurrentVersions(
AttributeError: type object 'VersionUpgradeManager' has no attribute 'getInstance'
```

sanderux, 2016-10-24:
My bad, needed to remove some python additions.
But Ultimaker 3 is still not showing, also not after removing config and cache

thopiekar, 2016-10-24:
That's strange. Could you show us the full log?

awhiemstra, 2016-10-24:
@probonopd Actually, I have some good news in that regard: http://software.ultimaker.com/current/Cura-2.3.0.AppImage

I actually wanted to have this posted as an experimental download on the website but that seems not to have happened. Feel free to share this around with anyone that is interested, as far as I know it works and is stable.

sanderux, 2016-10-24:
@awhiemstra thanks, that works on 16.04, however, i dont get any keyboard input in the appimage. Bit hard to change settings :)

@thopiekar here's the output
```
sander@dl01:~$ sudo apt-get install cura cura-extra-plugin-um3networkprinting 
[sudo] password for sander: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  cura cura-extra-plugin-um3networkprinting
0 upgraded, 2 newly installed, 0 to remove and 62 not upgraded.
Need to get 0 B/5405 kB of archives.
After this operation, 34,0 MB of additional disk space will be used.
Selecting previously unselected package cura.
(Reading database ... 310698 files and directories currently installed.)
Preparing to unpack .../cura_1%3a2.3.99-master~201610111546~rev1433~pkg32~ubuntu16.04.1_all.deb ...
Unpacking cura (1:2.3.99-master~201610111546~rev1433~pkg32~ubuntu16.04.1) ...
Selecting previously unselected package cura-extra-plugin-um3networkprinting.
Preparing to unpack .../cura-extra-plugin-um3networkprinting_2.3.99~201610231127~68~ubuntu16.04.1_all.deb ...
Unpacking cura-extra-plugin-um3networkprinting (2.3.99~201610231127~68~ubuntu16.04.1) ...
Processing triggers for bamfdaemon (0.5.3~bzr0+16.04.20160824-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for gnome-menus (3.13.3-6ubuntu3.1) ...
Processing triggers for desktop-file-utils (0.22-1ubuntu5) ...
Processing triggers for mime-support (3.59ubuntu1) ...
Processing triggers for shared-mime-info (1.5-2ubuntu0.1) ...
Unknown media type in type 'chemical/x-alchemy'
Unknown media type in type 'chemical/x-cache'
Unknown media type in type 'chemical/x-cactvs-ascii'
Unknown media type in type 'chemical/x-cactvs-binary'
Unknown media type in type 'chemical/x-cactvs-table'
Unknown media type in type 'chemical/x-cdx'
Unknown media type in type 'chemical/x-cdxml'
Unknown media type in type 'chemical/x-chem3d'
Unknown media type in type 'chemical/x-cif'
Unknown media type in type 'chemical/x-cml'
Unknown media type in type 'chemical/x-daylight-smiles'
Unknown media type in type 'chemical/x-dmol'
Unknown media type in type 'chemical/x-gamess-input'
Unknown media type in type 'chemical/x-gamess-output'
Unknown media type in type 'chemical/x-gaussian-input'
Unknown media type in type 'chemical/x-gaussian-log'
Unknown media type in type 'chemical/x-genbank'
Unknown media type in type 'chemical/x-gulp'
Unknown media type in type 'chemical/x-hin'
Unknown media type in type 'chemical/x-inchi'
Unknown media type in type 'chemical/x-inchi-xml'
Unknown media type in type 'chemical/x-jcamp-dx'
Unknown media type in type 'chemical/x-macromodel-input'
Unknown media type in type 'chemical/x-mdl-molfile'
Unknown media type in type 'chemical/x-mdl-rdfile'
Unknown media type in type 'chemical/x-mdl-rxnfile'
Unknown media type in type 'chemical/x-mdl-sdfile'
Unknown media type in type 'chemical/x-mdl-tgf'
Unknown media type in type 'chemical/x-mmcif'
Unknown media type in type 'chemical/x-mol2'
Unknown media type in type 'chemical/x-mopac-graph'
Unknown media type in type 'chemical/x-mopac-input'
Unknown media type in type 'chemical/x-mopac-out'
Unknown media type in type 'chemical/x-msi-car'
Unknown media type in type 'chemical/x-msi-hessian'
Unknown media type in type 'chemical/x-msi-mdf'
Unknown media type in type 'chemical/x-msi-msi'
Unknown media type in type 'chemical/x-ncbi-asn1'
Unknown media type in type 'chemical/x-ncbi-asn1-binary'
Unknown media type in type 'chemical/x-ncbi-asn1-xml'
Unknown media type in type 'chemical/x-pdb'
Unknown media type in type 'chemical/x-shelx'
Unknown media type in type 'chemical/x-vmd'
[... 144 lines cut ...]
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
file:///usr/share/cura/resources/qml/MonitorButton.qml:164: TypeError: Cannot read property 'jobState' of undefined
[... cut ...]
2016-10-24 13:30:47,661 - DEBUG - .../UM/Qt/GL/QtOpenGL.py (__init__ [19]): OpenGL Version:  3.0 Mesa 11.2.0
2016-10-24 13:30:47,662 - DEBUG - .../UM/Qt/GL/QtOpenGL.py (__init__ [19]): OpenGL Vendor:   X.Org
2016-10-24 13:30:47,662 - DEBUG - .../UM/Qt/GL/QtOpenGL.py (__init__ [19]): OpenGL Renderer: Gallium 0.4 on AMD CAICOS (DRM 2.43.0, LLVM 3.8.0)
2016-10-24 13:30:57,121 - DEBUG - /usr/lib/cura/plugins/AutoSave/AutoSave.py (_onTimeout [46]): Autosaving preferences, instances and profiles
```

And this is the printer selection available

[image]

thopiekar, 2016-10-24:
You are obviously using my master PPA :roll_eyes: 
Told this so many times: Don't use this PPA. There are blocked builds (eg. because of pull request which still need review here) and other possible bugs.
The stable PPA is tested and if needed I also apply dirty patches there to keep the packages built.

So please:
1. Remove the master PPA from APT.
2. Remove all local packages which have been installed from the master PPA.
3. Apply the stable PPA and reinstall everything from there.
4. _Most important_: Remove all settings and configs.
5. Run Cura.

Regards

sanderux, 2016-10-24:
@thopiekar how strange, i thought the ppa you linked *was* the stable.
I did
```
sudo add-apt-repository ppa:thopiekar/cura
sudo apt-get update
```

would that not give me the stable one?

sanderux, 2016-10-24:
how strange, i removed and re-added and now it does run with UM3

thopiekar, 2016-10-24:
Great 👌
Happy printing 😉

[44 more replies]

closed 2018-02-27 by ianpaschal