Ultimaker/Cura #719: error: program lacks a fragment shader

tomboy-64, 2016-04-23:
Cura builds fine on Gentoo, but does not display any 3d models. On the console I get the following output:
```$ cura
2016-04-23 22:19:22,576 - INFO - Loaded plugin ConsoleLogger
2016-04-23 22:19:22,614 - INFO - Loaded plugin CuraEngineBackend
2016-04-23 22:19:22,621 - INFO - Loaded plugin SelectionTool
2016-04-23 22:19:22,634 - INFO - Loaded plugin RotateTool
2016-04-23 22:19:22,645 - INFO - Loaded plugin ScaleTool
2016-04-23 22:19:22,650 - INFO - Loaded plugin TranslateTool
2016-04-23 22:19:22,657 - INFO - Loaded plugin CameraTool
2016-04-23 22:19:22,665 - INFO - Loaded plugin MirrorTool
2016-04-23 22:19:22,671 - INFO - Loaded plugin FileLogger
2016-04-23 22:19:22,673 - WARNING - Plugin ConsoleLogger was already loaded
2016-04-23 22:19:22,679 - INFO - Loaded plugin STLWriter
2016-04-23 22:19:22,684 - INFO - Loaded plugin STLReader
2016-04-23 22:19:22,731 - INFO - Loaded plugin 3MFWriter
2016-04-23 22:19:22,741 - INFO - Loaded plugin OBJReader
2016-04-23 22:19:22,748 - INFO - Loaded plugin OBJWriter
2016-04-23 22:19:22,791 - INFO - Checking for new version of cura
2016-04-23 22:19:22,791 - INFO - Loaded plugin UpdateChecker
2016-04-23 22:19:22,805 - INFO - Loaded plugin SimpleView
2016-04-23 22:19:22,814 - INFO - Loaded plugin WireframeView
2016-04-23 22:19:22,827 - INFO - Loaded plugin LocalFileOutputDevice
2016-04-23 22:19:22,837 - INFO - Loaded plugin PerObjectSettingsTool
2016-04-23 22:19:22,846 - INFO - Loaded plugin AutoSave
2016-04-23 22:19:22,860 - INFO - Loaded plugin SolidView
2016-04-23 22:19:22,872 - INFO - Loaded plugin LayerView
2016-04-23 22:19:22,884 - INFO - Loaded plugin ImageReader
2016-04-23 22:19:22,899 - INFO - Loaded plugin CuraProfileReader
2016-04-23 22:19:22,927 - INFO - Loaded plugin RemovableDriveOutputDevice
2016-04-23 22:19:22,939 - INFO - Loaded plugin GCodeProfileReader
2016-04-23 22:19:22,950 - INFO - Loaded plugin SliceInfoPlugin
2016-04-23 22:19:22,961 - INFO - Loaded plugin GCodeWriter
2016-04-23 22:19:22,961 - WARNING - Plugin CuraEngineBackend was already loaded
2016-04-23 22:19:22,972 - INFO - Loaded plugin 3MFReader
2016-04-23 22:19:22,986 - INFO - Loaded plugin ChangeLogPlugin
2016-04-23 22:19:22,999 - INFO - Loaded plugin LegacyProfileReader
2016-04-23 22:19:23,026 - INFO - Loaded plugin USBPrinting
2016-04-23 22:19:23,042 - INFO - Loaded plugin XRayView
2016-04-23 22:19:23,061 - INFO - Loaded plugin CuraProfileWriter
2016-04-23 22:19:23,063 - INFO - Command line arguments: {'file': [], 'disable-textures': False, 'external-backend': False, 'debug-mode': False}
2016-04-23 22:19:23,307 - WARNING - No active machine found when trying to save setting visibility
2016-04-23 22:19:23,310 - DEBUG - Attempting to kill the engine process
2016-04-23 22:19:23,479 - INFO - Started engine process: /usr/bin/CuraEngine
2016-04-23 22:19:23,683 - DEBUG - Backend connected on port 49674
2016-04-23 22:19:23,684 - DEBUG - Setting active view to SolidView
2016-04-23 22:19:23,994 - DEBUG - Calculating normals took 0.0006978511810302734 seconds
2016-04-23 22:19:23,995 - DEBUG - Loaded a mesh with 8052 vertices
2016-04-23 22:19:24,696 - DEBUG - Loaded theme /usr/share/cura/resources/themes/cura
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
2016-04-23 22:19:26,870 - DEBUG - Initialized OpenGL subsystems.
2016-04-23 22:19:26,870 - DEBUG - OpenGL Version:  OpenGL ES 3.1 Mesa 11.2.1
2016-04-23 22:19:26,870 - DEBUG - OpenGL Vendor:   Intel Open Source Technology Center
2016-04-23 22:19:26,871 - DEBUG - OpenGL Renderer: Mesa DRI Intel(R) HD Graphics 5500 (Broadwell GT2) 
QOpenGLShader::compile(Fragment): 0:7(1): error: No precision specified in this scope for type `float'
0:8(1): error: No precision specified in this scope for type `float'
0:12(1): error: No precision specified in this scope for type `float'
0:13(1): error: No precision specified in this scope for type `vec4'
0:15(1): error: No precision specified in this scope for type `vec2'
0:19(1): error: No precision specified in this scope for type `vec3'
0:20(1): error: No precision specified in this scope for type `vec3'
0:21(1): error: No precision specified in this scope for type `vec3'
0:29(1): error: No precision specified in this scope for type `vec4'
0:30(1): error: No precision specified in this scope for type `vec4'
0:34(1): error: No precision specified in this scope for type `vec4'
0:37(1): error: No precision specified in this scope for type `vec4'
0:41(1): error: No precision specified in this scope for type `vec4'

*** Problematic Fragment shader source code ***
#ifndef GL_FRAGMENT_PRECISION_HIGH
[... 70 lines cut ...]
file:///usr/lib64/qt5/qml/QtQuick/Controls/Menu.qml:149: TypeError: Cannot read property '__contentItem' of null
We got some errors while running testparm "Load smb config files from /etc/samba/smb.conf\nError loading services.\n"
KSambaShare: Could not find smb.conf!
2016-04-23 22:19:41,972 - DEBUG - Autosaving preferences, instances and profiles
2016-04-23 22:19:52,304 - DEBUG - Calculating normals took 0.004210472106933594 seconds
2016-04-23 22:19:52,304 - DEBUG - Loaded a mesh with 74712 vertices
2016-04-23 22:19:52,382 - DEBUG - Loading mesh took 2.572286605834961 seconds
2016-04-23 22:19:53,100 - DEBUG - Attempting to kill the engine process
2016-04-23 22:19:53,100 - DEBUG - Killing engine process
2016-04-23 22:19:53,113 - DEBUG - Engine process is killed. Recieved return code -15
2016-04-23 22:19:53,755 - INFO - Backend crashed or closed. Restarting...
2016-04-23 22:19:53,858 - DEBUG - Attempting to kill the engine process
2016-04-23 22:19:53,873 - INFO - Started engine process: /usr/bin/CuraEngine
2016-04-23 22:19:53,909 - DEBUG - Backend connected on port 49674
2016-04-23 22:19:54,495 - DEBUG - Sending data to engine for slicing.
2016-04-23 22:19:54,498 - DEBUG - Sending data to engine is completed
libEGL warning: FIXME: egl/x11 doesn't support front buffer rendering.
2016-04-23 22:20:20,493 - DEBUG - Backend quit with return code 0. Resetting process and socket.
2016-04-23 22:20:20,516 - INFO - Started engine process: /usr/bin/CuraEngine
2016-04-23 22:20:20,526 - DEBUG - Backend connected on port 49674
2016-04-23 22:20:21,975 - DEBUG - Autosaving preferences, instances and profiles
file:///usr/lib64/qt5/qml/QtQuick/Controls/Private/MenuContentItem.qml:178: ReferenceError: Acccessible is not defined
file:///usr/lib64/qt5/qml/QtQuick/Controls/Private/MenuContentItem.qml:178: ReferenceError: Acccessible is not defined
2016-04-23 22:21:06,128 - DEBUG - Shutting down cura
```

thopiekar, 2016-04-23:
Oh, that is strange. You are using OpenGL ES? I got that error message in the past when trying to run Cura on Ubuntu armhf. Are you able to use regular OpenGL w/ GLX?

tomboy-64, 2016-04-23:
I am.

thopiekar, 2016-04-23:
Well that's the problem then. There are missing instructions in the shader files.
I already played with one of them, see https://github.com/Ultimaker/Uranium/pull/117, and @awhiemstra gave me a tip how to do that better, but I don't get how all of that works, so I just left it there for now.

Maybe you can find a better way get it fully working :wink:

nallath, 2017-07-27:
Closing due to inactivity. I've not heard any more reports about this (and i've not been able to reproduce this). I'm assuming this is fixed.

closed 2017-07-27 by nallath

Ghostkeeper, 2017-08-09:
I suspect we still don't support OpenGL ES though...