Ultimaker/Cura #720: Better movement optimization.

rewolff, 2016-04-25:
Often the movements are not well optimized. A layer will be done in one direction leaving some very small patches to be done all over the place. 

Algorithmically: To optimize this:
Define: longmove: A move without extrusion longer than a threshold. 
Define: a patch: A sequence of extrusions without a longmove flanked by longmoves or a layer change. 

For each patch: 
find the closest shortmove and see if the total of the longmoves goes down if you insert this patch in the moves at that point. 

That would do a lot better already!

fieldOfView, 2016-04-25:
Pinging @BagelOrb

BagelOrb, 2016-04-29:
Pong @Ghostkeeper 

He already implemented this on a separate branch, but the suggested algorithm doesn't seem to benefit print times that much. Perhaps he can elaborate better.

Ghostkeeper, 2016-05-02:
It looks a lot like my [cluster algorithm](https://github.com/Ultimaker/CuraEngine/blob/PathOptimizer_RandomInsertion_fixed_history_2/src/pathOrderOptimizer.h#L104) (cluster = patch) but applied to closest-insertion algorithm rather than exhaustive random-insertion. The clustering serves to reduce the computational cost, and that works well.

That algorithm implementation, by the way, was made 7 months ago. I can hardly pretend to be the expert in optimising line order any more! I've heard some new ideas as well from other people to optimise the order. A lot can be done with various insertion methods, but all of those would need the current implementation overhauled.

rewolff, 2016-08-04:
Even if it is not the time that improves, the print quality is what matters to me. Whenever I get a finished print, I see the longmoves in the top surface that otherwise has a regular pattern.

Ghostkeeper, 2016-08-04:
Cura 2.2 has a feature to avoid the skin when combing. That prevents moves over the top layer like that.

closed 2018-11-26 by Appesteijn