Ultimaker/Cura #724: Request Setting to Disable retract in support

CCS86, 2016-04-27:
Some support areas generate quite a few retract moves. Generally, I'd just prefer them to be rapid moves without retract. Better yet allow coasting as well.

Ghostkeeper, 2016-04-28:
It should not retract when it's moving from support to support, except if there is a piece of model that intersects with the straight line from the start of the travel move to the end. It should (often) retract when moving from model to support or from support to model. That's the intended behaviour currently. Is it behaving differently?

I don't agree entirely with retracting if a straight line would intersect the model but Avoid Printed Parts is enabled (since the nozzle would move around the object anyway), but I heard it was deemed better that way.

CCS86, 2016-04-29:
Your last statement might be exactly what's happening. The move appears to clear any finished surfaces, yet a retract is called. You can imagine that with certain combinations of part geometry and support placement, you could end up with many unnecessary retracts. 

I'd love to hear the argument of why it's better this way, to see the whole picture. Maybe you can tag someone in that camp?

Ghostkeeper, 2016-05-03:
The argument is that if an oozing nozzle moves around an object, even with a wide berth, it might pull the oozed string against the surface of the model (like a lasso).
I have faith that the "other camp" knows it better than I, but @BagelOrb is the one that explained it to me.

mikenew12, 2016-05-04:
Maybe this should be a separate issue, but I really wish there was a setting to **enable** retraction between supports. I understand why it's off by default, but for certain prints, i.e. an artistic print with a complicated support structure, the stringing between supports can get so bad it ruins the print.

Jip-Hop, 2017-05-13:
I agree with @michaelnew. For the same reason I also need a setting to enable/disable retraction to work in combination with supports.

ianpaschal, 2018-02-27:
Although this thread is older, the duplicate has more discussion in it. Please see #1828 to continue this discussion.

closed 2018-02-27 by ianpaschal