Ultimaker/Cura #726: Cura 2.1 open beta: remember last used folder

Dim3nsioneer, 2016-04-28:
In the past Cura remembered the last used folder, e.g. when saving a gcode file. Today with the open beta it always starts on top folder which leads - depending on someones folder organization - to a painful browsing through subfolders everytime a file is save (gcode, profile) or loaded (stl, profile).
This is a feature request for making Cura remembering the last used folder again.

nallath, 2016-04-28:
For our reference; CURA-1497

fieldOfView, 2016-04-28:
I much agree with this request. Interestingly the "Save to File..." dialog does remember its state.
For our internal reference: CURA-1135

nallath, 2016-04-28:
Did we just made a duplicate issue? (if so, i'll delete mine)

fieldOfView, 2016-04-28:
CURA-1135 has been around for a bit longer.

Ghostkeeper, 2016-04-28:
Seems to be a regression.

MarcusWolschon, 2016-05-01:
I just reat in the Ultimaker Forum Thread regarding the 2.1 Beta that this is how Ultimaker decided it to behave. 
Sorry.
This annoys the crap out of me too. 
.. enough that I'm thinking about a fork with this change and otherwise kept up to date with git pulls.

thopiekar, 2016-05-01:
If I'm not wrong it is OS dependent. On Windows (and OSX?) you won't see that. For me on KDE (Linux) a legacy window is used by Qt5 which is very minimalistic and is just provided as a workaround (fallback) by Qt.
Yes, it is annoying that bugs like that always have to be blamed on 3rd party software (that's why cura-build is needed here to use software which is verified to work well), but that's where they actually come from.
@awhiemstra already reported that problem upstream at the KDE bugtracker, as it seemed to be KDE specific. Sorry, if I mixed here something up..

fieldOfView, 2016-05-02:
@MarcusWolschon if you do, could you please make a pull request out of it? We're all trying to make Cura the best it can be, but sofar the issue to make Cura remember the last used folder has been pushed forward because there are "bigger fish".

nallath, 2016-05-02:
If you do, just add a option to the preferences menu that allows you to enable / disable that feature. That would be the best option imho (yay for giving everyone what they want)

closed 2016-10-12 by BagelOrb