Ultimaker/Cura #728: [2.1 beta] Layer processing & displaying much slower than 15.04.05

claus1009, 2016-04-29:
1. Layer Processing speed of 2.1.beta is much much  slower than in 15.04.05.
2. In 15.04.05 during layer processing the portion of the object which has already been processed gets displayed. In 2.1.beta the platform stays empty until processing was finished by100%.
3. After processing was finished stepping through the layers with Ctrl+Arrow-Keys happens nearly instantuous and seamless in 15.04.05. But in 2.1. beta there is an extreme delay every time you are stepping from one layer to the next one (rolling arrow bisede the layer scrollbar & layer number indicator)

Layer display is really essential for me.
Although I like many of the new features of 2.1.beta for me this extreme degaradation in layer processing/displaying performace  really is a no go.
This makes working absolutly impossible.

So I returned to 15.04.05.

Info:
Ultimaker 2+
Intel i5-4210U
GeForce 840M

nallath, 2016-04-30:
The layer processing speed is indeed slower. Once they are loaded, the rendering is much faster though.

This won't be improved / changed for the 2.1 release.

closed 2016-04-30 by nallath

claus1009, 2016-04-30:
I can't  confirm that anything related to layer processing and displaying is faster than in 15.04.05.
Everything is unacceptably slower.
But OK it's your decission as the developer.
But please consider ...
You can implement the coolest new features and addons.
As long as you corrupt the basic user expirience in such a signifficant way, you do yourself and your developements no favour.
In my case ...
I stopped experimenting/testing with these new features, because I really don't have the luxury to spend all this extra-time waiting for layers to be processed and displayed.
If you really decide to leave layerprocessing and displaying the way it is now ... I can only make one clear statement ...
then for me personally 2.1.beta is a miscarriage!

daid, 2016-05-01:
We are aware of the issue, we are aware of the piece of code that made the 15.xx faster. However, when we tried to use that same code, it did not work and caused a lot of crashes.

So we opted for the slower more stable implementation for 2.1, instead of delaying the 2.1 release because of this regression.

claus1009, 2016-05-01:
daid! 
I'm fully aware that "time to realease date" sometimes requires uncomfortable "hop or drop" decisions.
And be assured ... if I wouldn't appreciate many of the new features (regarding shell, infill and especially the experimantal ones) of 2.1.beta so much I wouldn't have cared, just would have stepped back to an older version and wouldn't even have spent the time to give you feedback.

But I'm absolutely sure that **2.1.beta in principle is a tremendous step forward**.

But the layer-diplay is the entry-point where most of the users start experimenting and learning how to use/apply those new features.

**This is where the user experience starts.**

So I really want to give the honest feedback ... never opt for a compromise which corrups basic user experience. You really do yourself and your other developements no favour.

daid can you give a rough estimation, when (in which Version) it will be possible to test the new features without the layer-processing/display penalty.

Ghostkeeper, 2016-05-02:
I'm going to estimate 2.3 or 3.0 for your point 2. Just hazarding a guess though. Perhaps your point 1 can be fixed sooner.