Ultimaker/Cura #730: Can't set new "Materials"

Searider, 2016-05-01:
2.1 introduced the "Materials" drop down in the "Print Job" section but I can not find any way to add new materials to the drop down.

Thanks

fieldOfView, 2016-05-01:
For the 2.1 release, we chose feature-parity with 15.04. That also did not let you (easily) add materials. You may want to play around with the profiles that get installed with Cura to add materials, but that is not supported at this moment and may break in future versions.

closed 2016-05-01 by fieldOfView

Searider, 2016-05-01:
Ok. thanks

fieldOfView, 2016-05-01:
As a ps: what type of printer are you using?

Searider, 2016-05-01:
UM2+  (converted UM2)

fieldOfView, 2016-05-02:
That, unfortunately, is the most complicated printer to add a new material for at the moment :-(
For the UM2+, materials, nozzle sizes and quality profiles are mixed into one.

If you just want add a material for normal quality, 0.4mm nozzle, take the file .../resources/profiles/ultimaker2+/pla_0.4_normal.curaprofile and copy it to your cura profile folder "profiles" directory. Edit the file in a text-editor and change the line that says "material = PLA". Then start Cura. That should get you started.

Searider, 2016-05-03:
Thanks. I will give this a try.