Ultimaker/Cura #732: Flow isn

Raniz85, 2016-05-03:


AmedeeBulle, 2016-06-03:
This is because temperature is defined in the Material and the Material priority is higher than the Profile priority so the Material overrides.

I am struggling with that as well, but it is by design: the idea is to be able to use the same Profile with different Materials, where the Material properties override the Profile ones.

Note that if you don't like it this way, you can always create Materials without temperature, that way Profiles will stay untouched.

nallath, 2016-06-05:
We aren't very happy with how profiles work at the moment either. This is why we're reworking it as we speak.

The only reason why we did go forward with releasing it as it is, is because you have to make a release at some point. If you wait for everything to be perfect, you will never release anything.

closed 2016-06-05 by nallath