Ultimaker/Cura #734: Could not find resource ('default.shader',)

PauloMFerreira, 2016-05-05:
Hi,

Im getting this error on running Cura:

`2016-05-05 10:23:01,515 - DEBUG - Initialized OpenGL subsystems.
2016-05-05 10:23:01,516 - DEBUG - OpenGL Version:  2.1 ATI-1.32.25
2016-05-05 10:23:01,516 - DEBUG - OpenGL Vendor:   ATI Technologies Inc.
2016-05-05 10:23:01,516 - DEBUG - OpenGL Renderer: ATI Radeon HD 6490M OpenGL Engine
Traceback (most recent call last):
  File "/Users/PauloFerreira/Cura-git/Cura/UM/Qt/Bindings/MainWindow.py", line 177, in _render
    renderer.beginRendering()
  File "/Users/PauloFerreira/Cura-git/Cura/UM/Qt/QtRenderer.py", line 114, in beginRendering
    self._initialize()
  File "/Users/PauloFerreira/Cura-git/Cura/UM/Qt/QtRenderer.py", line 171, in _initialize
    self._default_material = OpenGL.getInstance().createShaderProgram(Resources.getPath(Resources.Shaders, "default.shader"))
  File "/Users/PauloFerreira/Cura-git/Cura/UM/Resources.py", line 70, in getPath
    raise FileNotFoundError("Could not find resource {0} in {1}".format(args, type))
FileNotFoundError: Could not find resource ('default.shader',) in 4`

Cura runs but I cant see the table area and cant import objects to the table.

Im using MacOSX.

fieldOfView, 2016-05-05:
It looks like you mixed the Cura and the Uranium repositories into one, is that correct? I don't think that is something we support. 

The idea is that you check out both the Cura and the Uranium repositories in their own folders. Then you make sure the Uranium folder is in the PYTHONPATH and you run python3 cura_app.py from the Cura folder.

fieldOfView, 2016-05-09:
@PauloMFerreira, did my comments solve your problem? I'm closing this issue for now, but if your problem persists let us know and I will reopen it.

closed 2016-05-09 by fieldOfView

PauloMFerreira, 2016-05-10:
Sorry for the late. You are right. Im mixed those two repositories. And your solution works fine.

Thanks a lot 👍

fieldOfView, 2016-05-10:
Thanks for confirming.