Ultimaker/Cura #735: Feature Request: Smooth layer transitions

thinkyhead, 2016-05-06:
I propose that instead of stopping XY and moving Z straight up in the transition between layers, use the last few mm of one layer do a "smooth transition" to the next layer (especially when printing perimeter-only objects, or transitioning in the perimeter.

This would be essentially the same as the "Joris" or "vase" style of printing, but just for the transitions. The slicer could account for the ramping up to the next layer with a gradually-increasing E flow, and a corresponding ramping-down of E flow when finishing that first perimeter (or infill).

This could potentially eliminate the "Z scar" altogether, by eliminating the stop where the extruder can ooze, and smearing out the material over more distance.

Does Cura already have this capability, but in some other form?

BagelOrb, 2016-05-09:
The problem is that the XY position you are left in at the end of a layer is generally not a part of the layer above.

Think of the last layer of the sitting piece of a chair. Doing a diagonal move towards the leaning piece would then smear a line across the whole sitting piece.

I don't think the z-scar is actually caused by moving up, but by the fact that you start and end your perimeter there...

thinkyhead, 2016-05-12:
Aha, good point. It would only work well with straight up-and-down walls.

BagelOrb, 2017-07-21:
We're probably not going to do this any time soon

closed 2017-07-21 by BagelOrb