Ultimaker/Cura #736: [2.1 Beta] Cura not printing to usb

tenten8401, 2016-05-07:
Cura 2.1 doesn't seem to recognize my Printrbot Simple.

thopiekar, 2016-05-07:
As you said, you already installed the needed drivers for your printer, I can only guess you are using Windows(?). Which version do you use?

tenten8401, 2016-05-08:
Windows 10 x64

Ghostkeeper, 2016-05-09:
If it's not registering the connection to the Printrbot Simple, the relevant code in Cura is here: https://github.com/Ultimaker/Cura/blob/master/plugins/USBPrinting/USBPrinterOutputDeviceManager.py#L223

This code was copied from here: https://github.com/daid/Cura/blob/SteamEngine/Cura/util/machineCom.py#L31

The only real difference is that the `serial_port_auto` setting is removed. The only thing I can see going wrong otherwise is that Cura 15.04 is still open and holding the COM port busy (or another instance of 2.1 for that matter).

tenten8401, 2016-05-09:
Didnt have 15.04 installed when doing the testing, but i didnt reboot my computer when i tried it. Will try it later after this 24h print finishes

nallath, 2016-05-09:
I ported / wrote most of that code, but as you might imagine, it was never tested on a printerbot. For the 2.2 there has been some refactoring, so you could try it again once we release that for beta.

awhiemstra, 2016-05-09:
In addition, there is a known issue on Windows 10. Microsoft decided that they wanted to include an arduino driver. However, this driver does not actually work but does conflict with the driver we supply. There is an article about it on our support website, but I cannot currently recall the link for it. You might want to verify that you are not running into that.

fieldOfView, 2016-05-09:
I think @awhiemstra is referencing this document:
https://ultimaker.com/en/resources/18627-change-arduino-drivers
Not sure if it is helpful, since 15.04 does recognize the printer

tenten8401, 2016-05-09:
After this print finishes, I will test with official arduino drivers and the windows 10 drivers, rebooting after each try. I will get back to you guys on whether it works or not. (Print's half way done!)

tenten8401, 2016-05-09:
After rebooting my computer, Cura still isn't recognizing the printer.
[image]

nallath, 2016-05-10:
Could you give us the logs? It should say a bit more there.

tenten8401, 2016-05-10:
All I can find are engine loss, where would I find the actual logs

nallath, 2016-05-10:
https://github.com/Ultimaker/Cura#logging-issues

tenten8401, 2016-05-10:
Will re-format my post later today.

tenten8401, 2016-05-10:
Here are the logs...
`2016-05-07 09:17:59,425 - INFO - Loaded plugin FileLogger
2016-05-07 09:17:59,443 - INFO - Loaded plugin GCodeProfileReader
2016-05-07 09:17:59,462 - INFO - Loaded plugin GCodeWriter
2016-05-07 09:17:59,506 - INFO - Loaded plugin ImageReader
2016-05-07 09:17:59,563 - INFO - Loaded plugin LayerView
2016-05-07 09:17:59,584 - INFO - Loaded plugin LegacyProfileReader
2016-05-07 09:17:59,614 - INFO - Loaded plugin LocalFileOutputDevice
2016-05-07 09:17:59,645 - INFO - Loaded plugin PerObjectSettingsTool
2016-05-07 09:17:59,676 - INFO - Loaded plugin PostProcessingPlugin
2016-05-07 09:17:59,709 - INFO - Loaded plugin RemovableDriveOutputDevice
2016-05-07 09:17:59,746 - INFO - Loaded plugin SliceInfoPlugin
2016-05-07 09:17:59,773 - INFO - Loaded plugin SolidView
2016-05-07 09:17:59,806 - INFO - Loaded plugin CameraTool
2016-05-07 09:17:59,857 - INFO - Loaded plugin MirrorTool
2016-05-07 09:17:59,901 - INFO - Loaded plugin RotateTool
2016-05-07 09:17:59,943 - INFO - Loaded plugin ScaleTool
2016-05-07 09:17:59,977 - INFO - Loaded plugin SelectionTool
2016-05-07 09:18:00,018 - INFO - Loaded plugin TranslateTool
2016-05-07 09:18:00,055 - INFO - Checking for new version of cura
2016-05-07 09:18:00,056 - INFO - Loaded plugin UpdateChecker
2016-05-07 09:18:00,125 - INFO - Loaded plugin USBPrinting
2016-05-07 09:18:00,166 - INFO - Loaded plugin SimpleView
2016-05-07 09:18:00,204 - INFO - Loaded plugin WireframeView
2016-05-07 09:18:00,247 - INFO - Loaded plugin XRayView
2016-05-07 09:18:00,248 - INFO - Command line arguments: {'disable-textures': False, 'external-backend': False, 'debug-mode': False, 'file': []}
2016-05-07 09:18:00,257 - DEBUG - Attempting to connect to COM3
2016-05-07 09:18:00,350 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-05-07 09:18:00,358 - DEBUG - Setting active view to SolidView
2016-05-07 09:18:01,309 - DEBUG - Loaded theme C:\Program Files\Cura 2.1\resources\themes\cura
2016-05-07 09:18:01,755 - DEBUG - Backend connected on port 49674
2016-05-07 09:18:01,953 - WARNING - Could not determine application version from string 2.1.0-, not checking for updates
2016-05-07 09:18:02,152 - INFO - Could not establish connection on COM3: 'Timeout'. Device is not arduino based.
2016-05-07 09:18:02,152 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 115200
2016-05-07 09:18:02,152 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 250000
2016-05-07 09:18:02,153 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 230400
2016-05-07 09:18:02,153 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 57600
2016-05-07 09:18:02,153 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 38400
2016-05-07 09:18:02,154 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 19200
2016-05-07 09:18:02,154 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 9600
2016-05-07 09:18:02,154 - ERROR - Baud rate detection for COM3 failed
2016-05-07 09:18:02,155 - DEBUG - Closing the printer connection.
2016-05-07 09:18:02,155 - WARNING - Printer connection state was not changed
2016-05-07 09:18:02,155 - WARNING - Printer connection state was not changed
2016-05-07 09:18:02,274 - DEBUG - Initialized OpenGL subsystems.
2016-05-07 09:18:02,274 - DEBUG - OpenGL Version:  4.5.0 NVIDIA 364.72
2016-05-07 09:18:02,274 - DEBUG - OpenGL Vendor:   NVIDIA Corporation
2016-05-07 09:18:02,274 - DEBUG - OpenGL Renderer: GeForce GTX 750 Ti/PCIe/SSE2
2016-05-07 09:18:12,531 - DEBUG - Autosaving preferences, instances and profiles
2016-05-07 09:18:12,531 - WARNING - No active machine found when trying to save setting visibility
2016-05-07 09:19:20,372 - WARNING - No active machine found when trying to save setting visibility
2016-05-07 09:19:21,268 - DEBUG - Attempting to kill the engine process
2016-05-07 09:19:21,268 - DEBUG - Killing engine process
2016-05-07 09:19:21,271 - DEBUG - Engine process is killed. Recieved return code 1
2016-05-07 09:19:21,704 - INFO - Backend crashed or closed. Restarting...
2016-05-07 09:19:21,769 - DEBUG - Calculating normals took 0.0005021095275878906 seconds
2016-05-07 09:19:21,769 - DEBUG - Loaded a mesh with 20250 vertices
2016-05-07 09:19:21,807 - DEBUG - Attempting to kill the engine process
2016-05-07 09:19:21,825 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-05-07 09:19:21,897 - DEBUG - Backend connected on port 49674
2016-05-07 09:19:22,391 - DEBUG - Attempting to kill the engine process
2016-05-07 09:19:22,391 - DEBUG - Killing engine process
2016-05-07 09:19:22,393 - DEBUG - Engine process is killed. Recieved return code 1
2016-05-07 09:19:22,622 - INFO - Backend crashed or closed. Restarting...
2016-05-07 09:19:22,725 - DEBUG - Attempting to kill the engine process
2016-05-07 09:19:22,800 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-05-07 09:19:22,895 - DEBUG - Backend connected on port 49674
2016-05-07 09:19:39,966 - DEBUG - Autosaving preferences, instances and profiles
2016-05-07 09:21:04,950 - DEBUG - Shutting down cura
2016-05-07 09:21:05,157 - ERROR - Exception while emitting shutdown signal: NotImplementedError('Stop should be implemented by subclasses',)
[... 1464 lines cut ...]
2016-05-09 19:57:11,134 - DEBUG - Setting value of speed_wall_x to 40 on profile Current settings
[... cut ...]
2016-05-09 19:58:11,392 - DEBUG - Sending data to engine for slicing.
2016-05-09 19:58:11,392 - DEBUG - Sending data to engine is completed
2016-05-09 19:58:12,221 - DEBUG - Backend quit with return code 0. Resetting process and socket.
2016-05-09 19:58:12,225 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-05-09 19:58:12,253 - DEBUG - Backend connected on port 49674
2016-05-09 19:58:14,923 - DEBUG - Autosaving preferences, instances and profiles
2016-05-09 19:58:33,821 - DEBUG - Autosaving preferences, instances and profiles
`

fieldOfView, 2016-05-10:
Looking at the logs, Cura tries and fails to communicate with a printer on the COM3 port. Do you know if your printer is on COM3, or on another port? Do you have any other serial devices connected to your computer?

tenten8401, 2016-05-10:
My printer is on COM3

thopiekar, 2016-05-10:
@fieldOfView I guess there is something wrong before he code tries to test the different baudrates.
```
2016-05-09 19:55:37,625 - INFO - Could not establish connection on COM3: 'Timeout'. Device is not arduino based.
```
For some reason this message worries me the most.

fieldOfView, 2016-05-10:
That's what I meant with "tries and fails to communicate with a printer on the COM3 port" ;-)

If COM3 is indeed your printer we have to find out why it is not responding to Cura.

tenten8401, 2016-05-10:
I have also tried removing "3D Builder" and disabling 3DPrintService with no luck.

tenten8401, 2016-05-10:
Currently the windows 10 Printrbot driver is installed, however I couldn't quite find what I should use as an alternative.

thopiekar, 2016-05-10:
@fieldOfView Ah, ok. Isn't there a tool like ```lsof``` on Windows?

thopiekar, 2016-05-10:
Ahh! That looks useful.

http://www.doublecloud.org/2014/05/which-process-holds-my-file-tools-on-linux-and-windows/

fieldOfView, 2016-05-10:
I don't think the serial port is in use by another process; it seems to be not responding (timeout) instead of saying it is busy.

Pinging @nallath, hoping he can help.

nallath, 2016-05-10:
The "not arduino based" error is spooned up when the ispBase can't connect to it. To be honest, I just copied that code from daid, as he warned me of the clusterfuck that is called USB printing. 

The lines below it:
2016-05-07 09:18:02,152 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 115200
2016-05-07 09:18:02,152 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 250000
2016-05-07 09:18:02,153 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 230400
2016-05-07 09:18:02,153 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 57600
2016-05-07 09:18:02,153 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 38400
2016-05-07 09:18:02,154 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 19200
2016-05-07 09:18:02,154 - DEBUG - Attempting to connect to printer with serial COM3 on baud rate 9600
2016-05-07 09:18:02,154 - ERROR - Baud rate detection for COM3 failed

Indicate that it correctly tried all baud rates, but the printer never gave a correct answer. Cura tries to call send a get-temp command. If it gets a number of correct responses it simply assumes that the current baud is correct (how else would we get a right answer...)

In this case, we never seem to get the right answer.  Note that some kind of connection was established (if it can't open the com port, we would expect to see that in the log as well)

tenten8401, 2016-05-10:
So basically, It's just the way Cura detects the printers and nothing I did wrong?

thopiekar, 2016-05-10:
I think so. It would be interesting to me how Cura acts in your case if you
are using Linux for example. Just to be 100%ly sure it us not printer
related, but related to Windows specific code.

Additionally you can take a serial console, called ```thermite``` or
something that sounds similar. I used that application years before to talk
with other serial devices. It was very handy.

There you can send the GCode (only the command to ask for the temperature
here) manually and see what your printer answers. It will be useful to
understand what's going wrong here.

tenten8401, 2016-05-10:
Okay.

thopiekar, 2016-05-10:
Just in case you are interested to provide extra information to find a
solution here 😉

tenten8401, 2016-05-10:
Downloaded something called Termite (not thermite) and got the following output from it:
`M105
ok T:25.9 /0.0 B:27.0 /0.0 T0:25.9 /0.0 T1:41.2 /0.0 T2:50.6 /0.0 @:0 B@:0`

thopiekar, 2016-05-10:
As reference this is what my bq Hephestos2 returns:
```M105```
```ok T:23.0 /0.0 B:0.0 /0.0 T0:23.0 /0.0 @:0 B@:0```

😉

tenten8401, 2016-05-10:
Probably why Cura isn't recognizing it at all.

thopiekar, 2016-05-10:
Seems so. We'll see tomorrow what the Ultimaker people will say about that. I'm sure we made a good job so far providing this info 😉

Good night ✌

Ghostkeeper, 2016-05-11:
Thanks for your dedicated feedback!
The difference is just that it reports temperatures for two additional nozzles, but it should be robust against that. Basically, the code just checks for the string `T:` in the response: https://github.com/Ultimaker/Cura/blob/master/plugins/USBPrinting/USBPrinterOutputDevice.py#L271

tenten8401, 2016-05-11:
That's odd, because my response does contain `T:`

daid, 2016-05-11:
Looks like some delay/timeout is missing somewhere, the old code would try each baudrate for at least 5 seconds. As you need to wait for bootloaders/startup time of the firmware. The log clearly shows that that is not happening.

@nallath did you test this code on Windows? Linux has far better serial port behavior then Windows. Meaning that if it works on Linux, it might fail on Windows.

nallath, 2016-05-11:
Uh. Didn't test it on windows, but there have been others that used it with an UM on windows if i recall correctly.

It could be that it's the combination of windows + printerbot firmware.

tenten8401, 2016-06-29:
This is still an issue on 2.1.2. Would love this fixed if possible.

nallath, 2016-06-29:
The USB printing has been tested on multiple platforms now and it works for Ultimaker machines. 

I don't have acces to  a printerbot (and Ultimaker definately won't buy me one ;)), so I can't fix / test this.

tenten8401, 2016-06-29:
I can do whatever you need done, collect serial output, whatever you need. Do you have skype or something? I would LOVE to get this fixed as the old Cura just looks like garbage compared to the new one.

thopiekar, 2016-06-29:
I'm sure that using socat (http://linux.die.net/man/1/socat) would be enough 😉
Looks simple and easy to use (at least when Linux is installed already), but never used it before.

nallath, 2016-06-30:
Are you able to run cura from source? In that case it's pretty easy for me to make a version for you that has a ton extra logging.

tenten8401, 2016-06-30:
I can do that.

nallath, 2016-06-30:
Could you send me a message at j.vankessel at ultimaker dot com?

I don't have time this week to help you with this, but i might be able to find some next week.

tenten8401, 2016-08-07:
@nallath think you could help me out with this? Haven't been on in a while. :P

thopiekar, 2016-08-07:
@tenten8401 Email sent to @nallath ?

nallath, 2016-08-09:
I've added more debug code to the USB auto detection stuff. Should be in tomorrows build.

fieldOfView, 2016-08-09:
(note that @nallath is talking about a build of Cura 2.2 that is not publicly available)

tenten8401, 2016-09-10:
Update: I may have found a potential problem! I attached to my serial port using CuteCom (Linux serial monitor), and sent M105. First output was "Active Extruder: 0". Send M105 again and got a valid temperature reading. Could this be why it's erroring out? Here's a video of the issue, I'm not sure why this is happening at all.

https://youtu.be/Rf0CzaIP3kA

fieldOfView, 2016-09-10:
If the firmware consistently answers "Active Extruder: 0" to the first M105 after a reset, that would indeed be a problem. Let's have a go at this theory.

In the folder to which Cura got installed, there should be a subfolder named plugins.
Open the file ```cura/plugins/USBPrinting/USBPrinterOutputDevice.py```
On or around line 337 it should read as follows:
```
                if b"T:" in line:
```
Change that to:
```
                Logger.log("d", "Response: %s" % line)
                if b"T:" in line or "Active Extruder" in line:
```
The first line is optional, but may help with debugging. Make sure the two lines have the same indentation, and uses spaces, not tabs. Keep a backup of the original just in case.

tenten8401, 2016-09-10:
Complete noob question here, but where would one find the Cura files when installing from the AUR? Full disk search is taking forever and I'm absolutely clueless. https://aur.archlinux.org/packages/cura/

fieldOfView, 2016-09-10:
I have no experience with Arch myself, so I am not sure. Candidates would be something like /opt/cura, /usr/bin/cura, /usr/local/bin/cura. But those are just guesses.

tenten8401, 2016-09-10:
Can't find it in /opt, /usr/bin/cura is a script, and /usr/local/bin/cura doesn't exist.

fieldOfView, 2016-09-10:
does the script have anything useful in it? ```cat /usr/bin/cura```

tenten8401, 2016-09-10:
Not really. http://hastebin.com/ixejigigam.py

fieldOfView, 2016-09-10:
I can keep guessing some more, but I think it is better if I wait until someone who knows linux chimes in. @thopiekar ?

thopiekar, 2016-09-10:
By default Cura goes to /usr/lib/python/dist-packages/cura.
Optionally it is possible that it goes to a directory with the python
version, which was being used during build, eg.
/usr/share/python3.5/dist-packages/cura.

Mit freundlichen Grüßen/With kind regards
- Thomas Karl Pietrowski

tenten8401, 2016-09-10:
Found it. /usr/lib/cura/plugins/USBPrinting/PrinterConnection.py .... I believe I'm using 2.1.2... which... aparently isn't right on your github releases page, as the AUR pulls it from 2.1.3.tar.gz

thopiekar, 2016-09-10:
@fieldofview also keep in mind that we are talking about Cura 2.1 here.
See on the AUR page that it is being used.

My upgrade to 2.3 is being held because of a strange bug during code import from GitHub (sadly).

thopiekar, 2016-09-10:
Oops, yes, that's where the plugins go to 😅

tenten8401, 2016-09-10:
For whatever peculiar reason, when adding that line to PrinterConnection.py, it magicially detects the printer and allows for printing to usb...

```
2016-09-10 12:42:06,739 - DEBUG - Response: b'echo:Unknown command: "~"\n'
2016-09-10 12:42:06,740 - DEBUG - Response: b'ok\n'
2016-09-10 12:42:06,742 - DEBUG - Response: b'ok T:28.6 /0.0 B:0.0 /0.0 T0:28.6 /0.0 @:0 B@:0\n'
2016-09-10 12:42:06,745 - DEBUG - Response: b'ok T:28.6 /0.0 B:0.0 /0.0 T0:28.6 /0.0 @:0 B@:0\n'
2016-09-10 12:42:06,747 - DEBUG - Response: b'ok T:28.6 /0.0 B:0.0 /0.0 T0:28.6 /0.0 @:0 B@:0\n'
2016-09-10 12:42:06,749 - INFO - Printer connection listen thread started for /dev/ttyACM0
2016-09-10 12:42:06,750 - INFO - Established printer connection on port /dev/ttyACM0
```

[60 more replies]

closed 2018-07-02 by ianpaschal