Ultimaker/Cura #738: Place Objects on table Algorithm

PauloMFerreira, 2016-05-11:
Hi,
I dont know if this is the right place to ask this.
I would like to locate and understand your algorithm of place the objects on the table.
I think that the algorithm is good ( center and no overlap function works fine) but have some limitations that I would like to try improve it.
Can you help me here?

Thanks

nallath, 2016-05-11:
What we currently do is actually pretty naive. We create 2D convex hulls of the objects (the "shadows") and use those to check if there is an intersection. If there is an intersection, the overlap is used to calculate a push free vector, which is used to move the object to a new location (which may or may not be free). 

All this logic happens in the PlatformPhysics class.

fieldOfView, 2016-05-11:
> (which may or may not be free)

And if it is not free, the dance continues.

PauloMFerreira, 2016-05-11:
Its pretty naive indeed. Cause it puts the first on center and the others up and down of that one. And never check if the space at right and left of the table is free. And I think that it has a bug: I put an object on the platform. If I right click on mouse and click "center object on platform" the convex hull dont cover all the object.

fieldOfView, 2016-05-11:
I would be surprised if it had only that one bug ;) We have a GIF somewhere where someone recorded an endless dance of ultimaker robots pushing eachother around.

Perhaps an approach that is not iterative would lead to a "stable" ordering quicker, but we're currently rather busy with other stuff. The convex hull generation is due for an update, but that may not be ready until after 2.2 I heard.

PauloMFerreira, 2016-05-11:
I thought that the algorithm was more developed and efficient. It works but not very well if I add more than 4/5 objects... Just a question: when I right click and do "center object on platform", the process is exactly the same when I upload a object to the platform?

nallath, 2016-05-12:
At the moment, yeah. There is ofcourse no set in stone reason to do it like that.

Ghostkeeper, 2016-05-12:
As far as I understood it basically just resets the translation component of the transformation matrix to 0. It counts as a move operation.

PauloMFerreira, 2016-05-12:
hum...right.
I looked at the stable release version (version 15.04) and the algorithm its pretty efficient.
If Im not wrong, cura 15.06 use that algorithm too, right?
But Im trying run that version and get this error:

`QWidget: Must construct a QApplication before a QWidget`

I could run cura-master so I have all Qt well installed. Maybe it needs a lower Qt version? I have installed Qt 5.6.

fieldOfView, 2016-05-12:
Cura 15.06 was the first (premature) release of the rewrite, and uses the same logic as 2.1. Please forget that 15.06 has ever existed ;-).

PauloMFerreira, 2016-05-12:
Oooh, crap :(
So...I have some way to get the old algorithm?

fieldOfView, 2016-05-12:
My guess would be all the way here:
https://github.com/daid/Cura/blob/SteamEngine/Cura/util/objectScene.py#L233

PauloMFerreira, 2016-05-12:
I think you are right.
Thanks a lot for the time and help.

👍

closed 2016-05-17 by PauloMFerreira