Ultimaker/Cura #739: [master]

thopiekar, 2016-05-14:
Based on the codebase from yesterday I noticed this when opening Cura.
The buildvolume of the ```Ultimaker 2``` is shown correctly.

[image]
[image]

fieldOfView, 2016-05-14:
Could it be related to https://github.com/Ultimaker/Cura/issues/717 ?

thopiekar, 2016-05-14:
Seems so. The only difference is that the contour lines are not flickering anymore.

thopiekar, 2016-05-14:
After a lot of time playing with the JSON file for my printer I couldn't find the reason there, but when removing ```~/.local/share/cura/*``` the problem just went away. Sorry, that I didn't made a backup of it to be able to debug it :disappointed:

closed 2016-05-14 by thopiekar

fieldOfView, 2016-05-14:
```
def on_new_issue(self, issue):
    if "buildvolume" in issue.subject and issue.printer.brand="bq" and issue.submitter == "topiekar":
        self._self_resolving_issues.append(issue)
        self.smile()
        return

    # \todo handle all other issues
```

thopiekar, 2016-05-14:
The code says everything :smile: