Ultimaker/Cura #743: [master] ABS is displayed in the list of materials, but the printer has no heated bed

thopiekar, 2016-05-14:
While removing the my settings I noticed that _ABS_ was taken as material. I fixed that by #742, but wouldn't it make more sense checking for:

```
"machine_heated_bed": {
            "default": false
        },
```

in the JSON files? Never played with ABS so far, but as far as I know there is no chance without heated bed.

nallath, 2016-05-17:
Yeah it would. We're reworking the settings at the moment, so we can take this into account right away.

closed 2016-05-17 by nallath

Ghostkeeper, 2016-05-17:
I think a printer should at least be allowed to print ABS without a heated bed. Maybe it has a heated chamber, for instance?

thopiekar, 2016-05-17:
Oh, well, good argument! But I would still make it optional.

* First of all, I don't expect everyone is interested in building that.
* Secondly,  vendors probably don't want that their printers are put into a chamber, so it is better that ABS doesn't appear there, so end-users don't come to that idea (at least not so quickly :sweat_smile: ).

So I think about a "show all materials" option in the preferences for the advanced users.
But I agree with @fieldOfView (https://github.com/Ultimaker/Cura/pull/742#issuecomment-219236537) that we should think about making PLA at least preferred.