Ultimaker/Cura #747: Cura 2.1.1 Beta crashes

Micr0SD, 2016-05-20:
Hi

Cura beta 2.1.1 crashes when opening stl file. Also menus are not readable, just blank boxes.
How to proceed?

Using Windows 7 Pro SP1
Hp Zbook14
Cpu - I5-4200U
Gpu - AMD Firepro M4100 FireGL V 
Ram - 8 GB

Best regards

Ghostkeeper, 2016-05-20:
Strange. We haven't seen anything like this on Windows 7.

Could you try deleting this folder: ``C:\Users\<Micr0SD's username>\AppData\Local\cura``

That should reset your settings to factory defaults.

Micr0SD, 2016-05-20:
2016-05-20 10:53:48,228 - INFO - Command line arguments: {'external-backend': False, 'disable-textures': False, 'file': [], 'debug-mode': False}
2016-05-20 10:53:48,839 - WARNING - No active machine found when trying to save setting visibility
2016-05-20 10:53:48,843 - DEBUG - Attempting to kill the engine process
2016-05-20 10:53:49,029 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-05-20 10:53:49,048 - DEBUG - Setting active view to SolidView
2016-05-20 10:53:49,132 - DEBUG - Backend connected on port 49674
2016-05-20 10:53:57,163 - DEBUG - Loaded theme C:\Program Files\Cura 2.1\resources\themes\cura
2016-05-20 10:54:00,758 - DEBUG - Initialized OpenGL subsystems.
2016-05-20 10:54:00,758 - DEBUG - OpenGL Version:  4.3.0 - Build 10.18.14.4332
2016-05-20 10:54:00,759 - DEBUG - OpenGL Vendor:   Intel
2016-05-20 10:54:00,760 - DEBUG - OpenGL Renderer: Intel(R) HD Graphics 4400
2016-05-20 10:54:15,220 - DEBUG - Autosaving preferences, instances and profiles
2016-05-20 10:54:20,447 - DEBUG - Calculating normals took 0.012000083923339844 seconds
2016-05-20 10:54:20,448 - DEBUG - Loaded a mesh with 9114 vertices
2016-05-20 10:54:20,467 - DEBUG - Loading mesh took 0.18401002883911133 seconds
2016-05-20 10:58:49,036 - INFO - Loaded plugin FileLogger
2016-05-20 10:58:49,069 - INFO - Loaded plugin GCodeProfileReader
2016-05-20 10:58:49,100 - INFO - Loaded plugin GCodeWriter
2016-05-20 10:58:49,144 - INFO - Loaded plugin ImageReader
2016-05-20 10:58:49,176 - INFO - Loaded plugin LayerView
2016-05-20 10:58:49,201 - INFO - Loaded plugin LegacyProfileReader
2016-05-20 10:58:49,229 - INFO - Loaded plugin LocalFileOutputDevice
2016-05-20 10:58:49,263 - INFO - Loaded plugin PerObjectSettingsTool
2016-05-20 10:58:49,298 - INFO - Loaded plugin PostProcessingPlugin
2016-05-20 10:58:49,339 - INFO - Loaded plugin RemovableDriveOutputDevice
2016-05-20 10:58:49,376 - INFO - Loaded plugin SliceInfoPlugin
2016-05-20 10:58:49,406 - INFO - Loaded plugin SolidView
2016-05-20 10:58:49,439 - INFO - Loaded plugin CameraTool
2016-05-20 10:58:49,478 - INFO - Loaded plugin MirrorTool
2016-05-20 10:58:49,524 - INFO - Loaded plugin RotateTool
2016-05-20 10:58:49,567 - INFO - Loaded plugin ScaleTool
2016-05-20 10:58:49,603 - INFO - Loaded plugin SelectionTool
2016-05-20 10:58:49,643 - INFO - Loaded plugin TranslateTool
2016-05-20 10:58:49,681 - INFO - Checking for new version of cura
2016-05-20 10:58:49,683 - INFO - Loaded plugin UpdateChecker
2016-05-20 10:58:49,763 - INFO - Loaded plugin USBPrinting
2016-05-20 10:58:49,803 - INFO - Loaded plugin SimpleView
2016-05-20 10:58:49,844 - INFO - Loaded plugin WireframeView
2016-05-20 10:58:49,888 - INFO - Loaded plugin XRayView
2016-05-20 10:58:49,889 - INFO - Command line arguments: {'disable-textures': False, 'external-backend': False, 'debug-mode': False, 'file': []}
2016-05-20 10:58:50,138 - WARNING - No active machine found when trying to save setting visibility
2016-05-20 10:58:50,141 - DEBUG - Attempting to kill the engine process
2016-05-20 10:58:50,254 - INFO - Started engine process: C:\Program Files (x86)\Cura 2.1\CuraEngine.exe
2016-05-20 10:58:50,280 - DEBUG - Setting active view to SolidView
2016-05-20 10:58:50,340 - DEBUG - Backend connected on port 49674
2016-05-20 10:58:51,770 - DEBUG - Loaded theme C:\Program Files (x86)\Cura 2.1\resources\themes\cura
2016-05-20 10:58:53,314 - DEBUG - Initialized OpenGL subsystems.
2016-05-20 10:58:53,314 - DEBUG - OpenGL Version:  4.3.0 - Build 10.18.14.4332
2016-05-20 10:58:53,314 - DEBUG - OpenGL Vendor:   Intel
2016-05-20 10:58:53,314 - DEBUG - OpenGL Renderer: Intel(R) HD Graphics 4400
2016-05-20 10:59:03,588 - DEBUG - Autosaving preferences, instances and profiles
2016-05-20 10:59:06,766 - DEBUG - Calculating normals took 0.0009999275207519531 seconds
2016-05-20 10:59:06,768 - DEBUG - Loaded a mesh with 9114 vertices
2016-05-20 10:59:06,782 - DEBUG - Loading mesh took 0.18001103401184082 seconds
2016-05-20 11:05:40,893 - INFO - Loaded plugin FileLogger
2016-05-20 11:05:40,916 - INFO - Loaded plugin GCodeProfileReader
2016-05-20 11:05:40,939 - INFO - Loaded plugin GCodeWriter
2016-05-20 11:05:40,967 - INFO - Loaded plugin ImageReader
2016-05-20 11:05:40,998 - INFO - Loaded plugin LayerView
2016-05-20 11:05:41,025 - INFO - Loaded plugin LegacyProfileReader
2016-05-20 11:05:41,055 - INFO - Loaded plugin LocalFileOutputDevice
2016-05-20 11:05:41,090 - INFO - Loaded plugin PerObjectSettingsTool
2016-05-20 11:05:41,134 - INFO - Loaded plugin PostProcessingPlugin
2016-05-20 11:05:41,177 - INFO - Loaded plugin RemovableDriveOutputDevice
2016-05-20 11:05:41,223 - INFO - Loaded plugin SliceInfoPlugin
2016-05-20 11:05:41,258 - INFO - Loaded plugin SolidView
2016-05-20 11:05:41,294 - INFO - Loaded plugin CameraTool
2016-05-20 11:05:41,340 - INFO - Loaded plugin MirrorTool
2016-05-20 11:05:41,411 - INFO - Loaded plugin RotateTool
2016-05-20 11:05:41,462 - INFO - Loaded plugin ScaleTool
[... 39 lines cut ...]
2016-05-20 11:11:58,656 - INFO - Loaded plugin SelectionTool
2016-05-20 11:11:58,688 - INFO - Loaded plugin TranslateTool
2016-05-20 11:11:58,721 - INFO - Checking for new version of cura
2016-05-20 11:11:58,721 - INFO - Loaded plugin UpdateChecker
2016-05-20 11:11:58,779 - INFO - Loaded plugin USBPrinting
2016-05-20 11:11:58,816 - INFO - Loaded plugin SimpleView
2016-05-20 11:11:58,851 - INFO - Loaded plugin WireframeView
2016-05-20 11:11:58,888 - INFO - Loaded plugin XRayView
[... cut ...]
2016-05-20 11:12:01,512 - DEBUG - OpenGL Renderer: Intel(R) HD Graphics 4400
2016-05-20 11:12:11,736 - DEBUG - Autosaving preferences, instances and profiles
2016-05-20 11:18:52,834 - DEBUG - Shutting down cura
2016-05-20 11:18:53,858 - ERROR - Exception while emitting shutdown signal: NotImplementedError('Stop should be implemented by subclasses',)
2016-05-20 11:18:53,859 - DEBUG - Attempting to kill the engine process
2016-05-20 11:18:53,859 - DEBUG - Killing engine process
2016-05-20 11:18:53,862 - DEBUG - Engine process is killed. Recieved return code 1

Ghostkeeper, 2016-05-20:
Cheers!

> 2016-05-20 10:53:49,029 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
> 2016-05-20 10:58:50,254 - INFO - Started engine process: C:\Program Files (x86)\Cura 2.1\CuraEngine.exe

It looks like you've tried both the 32-bit version and the 64-bit version.

@JorgM, one of our QA testers, mentions that there is a known runtime error in the 32-bit build of version 2.1. Perhaps that's what you are encountering here. But in later attempts to launch Cura you tried the 64-bit version and apparently got bugs there too.

Micr0SD, 2016-05-20:
Yes i tried both.. I copied last part of the logs. First i tried 64-bit version with bugs. Then i tried 32-bit version and finally stayed with 64-bit version.

After many restarts and uninstalls i got the software working. I don't know what caused the problem.

Ghostkeeper, 2016-05-23:
Just thinking out loud here.

> 2016-05-20 11:11:59,114 - WARNING - No active machine found when trying to save setting visibility

Maybe that has something to do with it...

fieldOfView, 2016-05-23:
No, that can't be it. That is one of those "this always happens and is nothing to worry about" warnings.

closed 2016-12-26 by thopiekar