Ultimaker/Cura #749: Cura 2.1.1 placing / rotating models might cause endless model translation

Spirou42, 2016-05-20:
Hi, while trying to slice http://www.thingiverse.com/thing:584714 cure got confused and endlessly tries to position the model inside the build chamber. This process of positioning and repositioning might have 3 endings.

1. after a couple of repositioning sequences cura stops and the model is off screen 
2. after a couple of repositioning sequences cura stops and the model is on the build plate
3. cura never stops and the positions oscillates between two incorrect positions.

have a look at https://youtu.be/isWzPVfWhvY for case 2 and 3

Ghostkeeper, 2016-05-23:
Hmm, something seems up with the bounding box calculations...

mokusolo, 2016-12-27:
Would you please tell me how you optimize this?
I haven't met this problem with Cura 2.3.Maybe you've revised it?
My friend used to try to read stl file and show view with vc. He has the problem too. It just run everywhere.

Ghostkeeper, 2016-12-27:
We had fixes in 2.3.0 that revised this code thoroughly. Then we also had fixes in 2.4.0 that tweaked it a bit further. We don't seem to have this problem in 2.4 any more, or I haven't seen it yet anyway.

We also have a research project going on to see if we can make the platform collision completely anew (for the better) with an actual physics engine.

nallath, 2017-07-27:
This is no longer an issue.

closed 2017-07-27 by nallath