Ultimaker/Cura #750: don't send any user data through the internet by default

greyltc, 2016-05-24:
Hey, It looks like by default [cura now sends *whatever it does* through the internet to https://stats.youmagine.com/curastats/slice](https://github.com/Ultimaker/Cura/blob/master/plugins/SliceInfoPlugin/SliceInfo.py). Please turn this off by default.  
The warning banner telling the user that her info is being leaked is not sufficient.

fieldOfView, 2016-05-24:
Though I personally don't like the sending of information to youmagine.com any more than you do, it is nothing new. Legacy Cura does the same thing, but it does not tell you about it.

Note that youmagine is a ultimaker "project", so we are not sending the info to a 3rd party. The info is used to analyse what features are actually being used, so we can see where best to "spend our money".

I am afraid this is something we cannot do too much about since people "higher up"  want it to be an opt-out instead of an opt-in. And they pay our bills.

closed 2016-05-24 by fieldOfView

greyltc, 2016-05-24:
Thanks for speaking candidly @fieldOfView. That's too bad to hear this was previously being transmitted in secret! I feel betrayed.

Maybe someone should explain to those that pay the bills that especially maker-type people (Ultimaker's customers) don't like this kind of info leaking and there could be some backlash now that there's a warning banner which makes this practice known. Keeping this on by default is a great way for Ultimaker to toss out the trust of their customer base; that should be weighed carefully against whatever gain Ultimaker gets from the data it's collecting from cura.

nallath, 2016-05-25:
Uh. Thats not true FieldOfView! 

It did mention this when first starting legacy Cura. I'm a 100% sure about this. It asks you about this in the first run wizard. This means that we've been collecing this info for at least two years and have asked everyone if they want this. 

I don't really think there is that big of an issue here. Anyone can see what information we send. We have at points also shared what information we have recieved in this way. The -only- information we send is a hash of the model (so we don't even know what model you are printing) and the settings you used to do it. We don't collect any information that can identify you in any way!

fieldOfView, 2016-05-25:
My bad. I threw out my legacy Cura profile and restarted it, and must have missed the opt out checkbox below the add machine wizard. Still, it is turned on by default (and apparently easy to miss)

A thing to note is that currently we don't even send that model hash:
https://github.com/Ultimaker/Cura/blob/master/plugins/SliceInfoPlugin/SliceInfo.py#L104