Ultimaker/Cura #751: M163/M164 generates useless/abusive errors in the console (layer view)

MoonCactus, 2016-05-24:
I know M163 / M164 are only supported by Repetier host so far but they are working and documented on http://reprap.org/wiki/G-code#M163:_Set_weight_of_mixed_material

I wrote a plugin to mix colors for the diamond hot end (among others), and Cura generates thousands of lines like these:
```
Unknown M code:163
Unknown M code:164
...
```
It probably could be silenced instead without danger (or written once only, since there are no line numbers associated with the warning).

fieldOfView, 2016-05-25:
Is this for Cura 15.4.x (and older) or for Cura 2.1?
Issues for Cura 15.04 should be posted here: https://github.com/daid/cura

MoonCactus, 2016-05-25:
Sorry, actually I did not even realize there was a significant new fork at Ultimaker. The version numbering may get a bit misleading imho (just as Java)...
I will re-post into David's legacy version (15.04 indeed), thanks!

closed 2016-05-25 by MoonCactus

Ghostkeeper, 2016-05-25:
Probably also a probleem on newer versions though.