Ultimaker/Cura #753: plugin: OctoPrint upload

markwal, 2016-05-25:
I'm working on a new version of OctoPrint upload with settings UI to set octoprint instance address, api_key, etc.  I hope you don't mind using the github issue tracker for a few dev questions.  Would you prefer the ultimaker forums?

Would you be interested in a PR to include such a plugin in a future version?

If I want to use a pypi package like requests with it, what is the Ultimaker way to do that?

Or should I use PyQt5.QtNetwork.QNetworkAccessManager instead of requests?

fieldOfView, 2016-05-25:
That would be awesome. I have been aching to do it, but we've been a little swamped with work.
There will be some changes "soon" that would make it much easier to do this in version 2.2 and beyond. Perhaps @nallath can give you some pointers (including what packages to use going forward).

nallath, 2016-05-25:
1. Not a problem. You could also use the forum if you prefer that, but I check both.
2. Yes :) 
3. I'd use QNetworkAccessManager if I were you. The Qt network stuff is in c++, so for large uploads you don't grab the GIL of python (which causes the interface to freeze).

markwal, 2016-05-26:
OK.  Giving it a whirl.  Thanks for the info.

closed 2016-05-26 by markwal