Ultimaker/Cura #754: Cura 2.1.x on Linux 32bit

smiling-Jack, 2016-05-26:
Hallo
why there is no build for a 32 bit Linux (Ubuntu) mashine ? it is not posible ? can I make my own 32bit build ? (perhaps with the build script ?)

Please, I want test Cura 2.1 but i only have 32bit systems in my workshop

nallath, 2016-05-26:
Because hardly anyone uses 32 bit linux machines, so there is also little support from libraries that we depend on.
Aside from that; Cura tends to need quite a bit of RAM, so 32 bits isn't the best way to get the most out of Cura.

thopiekar, 2016-05-28:
@smiling-Jack You can find i386 builds here: https://launchpad.net/~thopiekar/+archive/ubuntu/cura
Don't forget that this PPA is not maintained by Ultimaker, so please blame me first when problems occur.

closed 2017-07-27 by nallath

4ndyD, 2021-10-04:
@thopiekar thanks for your work on this! I'm trying to set up an old eeePC 701 as a print controller, but after trying and failing to install Cura from your PPA I read on your website that the minimum supported OS is U18.10, whereas the last OS I could get supporting i386 was Lubuntu 18.04, and even that had some problem getting the right video resolution, so I've got it back running on 16.04 at the moment.
Do you know if there's any way still to get _any_ version of Cura running on such an old system?