Ultimaker/Cura #755: Crashes - windows 10 - incorrect syntax (2.1.1)

clearbucketLabs, 2016-05-27:
This is the log from 2.1.1, i initially installed it on D: but then on C: (after uninstall) but the error is the same.  Also 15.04 does not start either, maybe the same issue.  15.04 just silently closes.   However, it's important to note that i don't even have a C:\Python34 Folder... so not sure what it's referencing here. 

Traceback (most recent call last):
  File "cura_app.py", line 25, in <module>
  File "C:\Python34\lib\os.py", line 227, in makedirs
  File "C:\Python34\lib\os.py", line 227, in makedirs
  File "C:\Python34\lib\os.py", line 227, in makedirs
  File "C:\Python34\lib\os.py", line 227, in makedirs
  File "C:\Python34\lib\os.py", line 227, in makedirs
  File "C:\Python34\lib\os.py", line 237, in makedirs
OSError: [WinError 123] The filename, directory name, or volume label syntax is incorrect: '\\D:'

Ghostkeeper, 2016-05-27:
Huh. We package our Python code with Py2exe, which (as far as I know) should include the Python interpreter in the binary itself, and not refer to C:\Python34...

I guess my best advice would be to try to install Python on your computer. You can get it [here](https://www.python.org/downloads/release/python-344/). Cura ships with version 3.4, but you could venture to try version 3.5.1 because it's the newest. Installing Python should at least make sure that the link to Python is clear to the computer, because it places the Python executable on your %PATH% environment variable. Since this error looks to have something to do with directories, that could be a workaround for you.

Cura 15.04 used Python 2, so that might be unrelated.

fieldOfView, 2016-05-27:
Could you try removing C:\Users\[yourname]\AppData\Local\cura and running Cura again? When uninstalling Cura, the configuration is not removed. Perhaps your config got corrupted somehow (since the reference to D: is still there)

clearbucketLabs, 2016-05-27:
@Ghostkeeper there is no error related to not finding python, so it's not a python issue.   Just that that log has that path, which doesn't exist.  most likely an internal alias of some sort.

@fieldOfView  There is no folder there.  I've looked already.  Even when it's installed it's not there.  However regardless the error shows '\\D:' which isn't valid anyways, it should be D:\\  at least, unless it was supposed to be escaping it somehow.  it should be able to install anywhere on any drive, so it's some bug or some other software on the system messing it up.   its a fresh windows 10 install, with the exception of source tree and visual studio being installed...  i will keep looking for the config unless it's in the registry.

clearbucketLabs, 2016-05-27:
Given that it seems the problem is it cannot create the config folder in the profile, it might be some path/environment variable messing  it up on my end.

clearbucketLabs, 2016-05-27:
Yeah, that was it... there was a "HOME" environment variable which shouldn't of looked at anyway that was a specific syntax for another app.

closed 2016-05-27 by clearbucketLabs