Ultimaker/Cura #756: 2.2.1: Z Shift in Model Footprint

CCS86, 2016-05-27:
I'm not sure if this is just a graphical glitch, or more serious issue. After importing this model I see:

[image]

[HTC M9 Lens Cover.stl.txt](https://github.com/Ultimaker/Cura/files/286896/HTC.M9.Lens.Cover.stl.txt)

fieldOfView, 2016-05-27:
Have you dragged to object around at all? It is possible to push an object into the buildplate to cut off a portion of the bottom. If you drag an object up, it should fall back onto the buildplate again.

CCS86, 2016-05-27:
That is, as-loaded. Dragging doesn't change it, nor does "lay flat". The model's frame is coincident with the XY plane.

fieldOfView, 2016-05-28:
I see, it is an extremely small object, so you are zoomed in quite a bit.
I am 98% sure this is a graphics glitch. The "shadow" is drawn slightly above the buildplate to prevent graphic artefacts due to "depth fighting" (https://upload.wikimedia.org/wikipedia/commons/b/bb/Z_buffer-Z_fighting.png). Normally you don't see that, but when you are looking from very close-by it may become visible.

closed 2016-05-28 by fieldOfView

Ghostkeeper, 2016-05-30:
The shadow is 0.1mm above the build plate to reduce z-fighting. Ideally we'd incorporate the shadow in the texture of the build plate but that has all sorts of difficulties.

If you zoom out all the way, you can still see the z-fighting, by the way. At some point the floating point precision in the depth image becomes too coarse.