Ultimaker/Cura #757: [master-472012a] Hidden settings icon went wonky

markwal, 2016-05-28:
Steps:

1. Add a printer (I chose Ultimaker Original)
2. Choose Advanced setup

First result:

Note all the settings categories have the info icon that gives the following message on hover:

"Some hidden settings use values different from their normal calculated value.  Click to make these settings visible."

Steps continued:
3. Click on the icon

Result:

Nothing happens

Expected:

To see the icon only where there are hidden values and when clicked, for them to show up.

fieldOfView, 2016-05-28:
Good catch, but a known issue. 

Master will be a bit wonky the next couple of weeks after a rather big merge last week. 

For a more stable experience try the 2.1 branch or a commit from before last thursday (make sure you get a matching Uranium and CuraEngine too)

markwal, 2016-05-28:
Nah, don't want the stable experience.  Trying to work on the OctoPrint Upload plugin, so I want to trudge through latest.  Just thought I'd note what I see, but I'll hold off entering stuff for the moment since the big settings merge seems to have broken quite a bit.  (I can't seem to be able to turn on support generation, for example.)

closed 2016-05-28 by markwal