Ultimaker/Cura #758: Show Material Usage in Grams

jakepoz, 2016-05-30:
Cura should show print material usage in grams. Filament is sold by weight, with spools sold as 750g, 1kg, or even by the imperial pound. It's not common knowledge to know how to convert between weight and meters of filament. And this calculation will be dependent on the material and diameter. Listing this by weight is more clear and works for all materials and printers.

Old way
[image]

Better way
[image]

Ghostkeeper, 2016-05-30:
There is a plug-in that gives this functionality: https://github.com/nallath/PrintCostCalculator

Also, for version 2.2, we're including this functionality by default.

fieldOfView, 2016-05-30:
Cura knows exactly how "long"/"far" it tells the 3d printer to extrude. To calculate the weight, we need to know the filament diameter and the density of the material. The former is known to a fair accurate value, the latter is very material-dependent. There would have to be a way to set that property. We're working on adding that

DDDirk, 2016-05-30:
Note that prints are often not fully dense, depending on the settings (notably extrusion speed) and the printer (there should be improvement with the UM2+). So the actual weight of the model and the length of extruded filament will often be lower than calculated.

BerndJM, 2016-05-30:
> There would have to be a way to set that property
Aren't the material profiles intended to store such information ?!?

fieldOfView, 2016-05-30:
> Aren't the material profiles intended to store such information ?!?

Eventually: yes. 
Currently: no.

Anyhow, the figure in length of material is always going to be more accurate than the figure in weight (or cost!).

thopiekar, 2016-05-30:
@DDDirk I guess for these cases a correction factor would be useful in 2.2. Just a simple multiplier in the general settings or printer specific as an hidden option.

DDDirk, 2016-05-31:
Well, that depends not only on the print speed and the printer (and if the nozzle is (slightly) clogged for example), but also on the material (even the colour can make a big difference) and still other factors, so it's not that straightforward. I am researching this in different ways, but until such things are more certain one should keep in mind that the value a slicer gives for the extruded material (be it in length or weight) is a nominal, calculated, value, which may differ greatly from the actual value (especially at high extrusion speeds).

Ghostkeeper, 2016-05-31:
The calculation it makes is a naive one: pi * r^2 * extruded length * density

thopiekar, 2016-05-31:
@DDDirk I understand what you mean. By adding colors the mixed PLA filaments changes it's properties, like viscosity, melting point, etc.
I just thought about a simple way to correct it, like a multiplier used when calculating a general overhead surcharge.
That's true! E.g the time is also a estimated value. The G-code does not tell how much time an action take, but tells where the action ends. I guess the estimated time is calculated via constant speed and does not take care about the mass inertia, etc.

@Ghostkeeper That brings me to an idea. What about adding a tooltip, which tells the user about that fact? For us it is trivial, but new endusers might expect something more precise.

BerndJM, 2016-05-31:
For me all this calculations about printtime and used filament is not (and must not be) a exact 	science rather a kind of indication? (don't know the right word for "Anhaltspunkt").
For example the printtime calculation in the legacy cura didn't ever meet the real performed printtime on my machine (because cura doesn't know my firmware settings) so I tweak the code to multiply the calculated time with a correction factor and this went very well.
Also the filament usage in m and gr are only estimated but I see rough if the filament I have is enough for the planned printjob. Surely I didn't start a job if cura says it needs 99grams of material and I have only 105grams left on my spool ;-))

jakepoz, 2016-05-31:
Yes I agree it certainly doesn't need to be exact, and I'm glad it's coming back in the next version :) If it's just an estimate, it's a more useful estimate if it's in the same units as filament is sold in.

MarcusWolschon, 2016-07-05:
Is there a material selection somewhere in Cura that could be a starting point to implement this?
I found the directory with the material profiles but no UI control for the user to ever select a material.

fieldOfView, 2016-07-05:
@MarcusWolschon: what type of printer do you use? For the UM2, it was long ago decided that material was to be handled on the printer, which is why there is no material selection in Cura if you select a UM2

MarcusWolschon, 2016-07-05:
Yes, a UM2 with the modification has_materials=true and I also created an i3 Prusa for testing.
(I'm currently still working on creating a UM2 profile for OctoPrint. Ulti-GCode doesn't work there, so material must be handled in the Slicer.)

fieldOfView, 2016-07-05:
For use with octoprint, you should set the gcode flavor to reprap and add start- and end-gcode:
https://gist.github.com/fieldOfView/ae85f755fc70d1c65e6968a870dfef0c
(this is for 2.1; for 2.2 the syntax different)

fieldOfView, 2017-04-27:
This issue can be closed

closed 2017-05-01 by nallath