Ultimaker/Cura #761: [2.1.1] Segfault on Debian Jessie

nickthetait, 2016-06-02:
1) Download 2.1.1 from https://ultimaker.com/en/products/cura-software/list
2) Install with `sudo dpkg -i Cura-2.1.1-Linux.deb`
3) Run with `cura`

Results: segfault with the following output...
`/usr/bin/cura: line 3:  9336 Segmentation fault      LD_LIBRARY_PATH=/opt/cura/lib PYTHONPATH=/opt/cura/lib/python3/dist-packages QT_PLUGIN_PATH=/opt/cura/lib/plugins QML2_IMPORT_PATH=/opt/cura/lib/qml QT_QPA_FONTDIR=/opt/cura/lib/fonts python3.4 /opt/cura/bin/cura "$@"`

nickthetait, 2016-06-02:
Already figured out what went wrong, just want to leave a note here for other people that run into this problem. The fix is to install `python3-pyqt5`

thopiekar, 2016-06-02:
Doesn't make sense. Ultimaker .deb comes with PyQt5 precompiled, isn't it? 🤔

nickthetait, 2016-06-03:
Tried with a fresh install of Debian Jessie inside a VM.
Tried using Cura 2.1.1 without installing anything else prior gave this error:
[image]

So I installed `python3-pyqt5` & re-installed Cura. It then ran just like normal. So it seems like this package should be added into the build process. FYI for @awhiemstra

Ghostkeeper, 2016-06-05:
This thread on StackOverflow may give a solution: https://stackoverflow.com/questions/17106315/failed-to-load-platform-plugin-xcb-while-launching-qt5-app-on-linux-without

thopiekar, 2016-06-06:
Seems to be the same solution like I've done for the NVidia problem.

The only question: Is "libqxcb.so" always there? Or is it an optional module?

nickthetait, 2016-06-07:
Just experienced the same using Cura 2.1.2
1) Fresh install of Debian Jessie inside VM
2) Install Cura
3) Run Cura, error message as in screenshot earlier
4) Install `python3-pyqt5` 
5) Run Cura just fine :smile:

nallath, 2016-06-08:
Huh. That is odd. @awhiemstra any takes on this?

awhiemstra, 2016-06-08:
I have no clue. I do not know enough about debian specifically what causes it.

thopiekar, 2016-06-08:
A answer may give 'strace -e open <cura_script>'. It will list all files
being opened and also show which PyQt5 is being used.

nickthetait, 2016-06-09:
1) Fresh install of Debian Jessie inside VM
2) Install strace
3) Install Cura 2.1.2
3) Run `strace -e open /usr/bin/cura`
4) Get lots of juicy debugging output:
[image]

Not sure if this is quite enough information. Let me know if you need something else.

thopiekar, 2016-06-09:
And with python3-pyqt5 installed?

nallath, 2017-07-27:
Closing due to inactivity. This seems to no longer occur as we didn't have any new mentions of this.

closed 2017-07-27 by nallath