Ultimaker/Cura #763: 2.1.1 not tagged as release

thopiekar, 2016-06-04:
When going to https://github.com/Ultimaker/Cura/releases the 2.1.1 codebase is not available to download.

Could you please tag this release?

MarcusWolschon, 2016-07-05:
2.1.2 isn't tagged as added as a release too.

https://github.com/Ultimaker/Cura/tags

thopiekar, 2016-07-05:
That's correct 😑

nallath, 2016-07-05:
We're very good at forgetting things :$

CReimer, 2016-07-07:
The same goes for CuraEngine, Uranium and libArcus.
Well, for libArcus it's actually worse. There isn't even 2.1.0 tagged.

TZanke, 2016-07-08:
I think this issue is related to https://github.com/Ultimaker/cura-build

ubuntu-15.04-build-script.sh should have an optional option like "--version 2.2.0".
If argument is set, create tag at the end of script and push tags to github.

Optionally, if version is set, upload created files to download section. (very basic deployment process)

Best regards
Tobias

Ghostkeeper, 2016-08-05:
2.1.3 is tagged.

closed 2016-08-05 by Ghostkeeper