Ultimaker/Cura #766: [2.1.1] 'Enable Cooling Fans' is disregarded if child properties are changed

AmedeeBulle, 2016-06-05:
If you change a fan property (like setting 'Regular Fan Speed' to 50%) and then uncheck the 'Enable Cooling Fans', you still get fan in your print...

E.g. in my GCode
```
...
;LAYER:1
M106 S14.167
G0 X117.934 Y61.715 Z0.500
...
```

(I've seen several issues reported with fan, but it is unclear if this one is covered, feel free to close if it is)

Ghostkeeper, 2016-06-05:
Haven't seen this one yet.

I'm guessing this is in the engine. Looks like it just always writes a fan speed regardless of the fan being enabled or not: [here](https://github.com/Ultimaker/CuraEngine/blob/5878916b03439922d227beff33d12e163ffc326f/src/gcodePlanner.cpp#L561) and [here](https://github.com/Ultimaker/CuraEngine/blob/5878916b03439922d227beff33d12e163ffc326f/src/gcodeExport.cpp#L730). The only exceptions are Makerbot printers and when fan speed is set to 0 (and when fan speed is not changed from a previous piece).

nallath, 2017-07-27:
Just tried this, doesn't seem to occur anymore.

closed 2017-07-27 by nallath