Ultimaker/Cura #768: Alternate Extra Wall + Spiralize Outer Contour

brazenrain, 2016-06-05:
Alternate Extra Wall adds material when Spiralize Outer Contour is active, even though infill is disabled.

Ghostkeeper, 2016-06-05:
It's a question whether this is undesired behaviour or not.

On the one hand, Alternate Extra Wall is intended to make the wall stick better to the infill. If there is no infill, that intended use is gone.

On the other hand, the functionality of Alternate Extra Wall is to add an extra wall on every other layer. You ask for an extra wall on every other layer, you'll get one. Perhaps there are other reasons you want that extra wall, such as for strength, or just that ribbed feeling for her pleasure.

I'm siding with the other hand, though it's really just a question of ease of use (automatically disable Extra Wall if it's not needed) vs. degree of control (do as the user asks).

BagelOrb, 2017-03-08:
That's a bit weird behaviour indeed.
Shouldn't happen in Cura 2.5 any more. In the new Cura only one polygon is printed.

closed 2017-03-08 by BagelOrb