Ultimaker/Cura #769: Rotating and Scaling are not commutative

brazenrain, 2016-06-05:
Rotating a part before scaling it causes the inputs for scaling to go haywire.

sedwards2009, 2016-06-05:
There are a bunch of problems related to the tools and especially rotating and scaling. A heap of them have been solved in the `simple_convex_hull` branches for Cura and Uranium. The Cura team is really busy for the next week or two and after that we will merge this branch.

luukakkerman, 2016-06-17:
Just to confirm this issue: Rotating an object makes the displayed scaling info on X, Y and Z go wild.
The resulting values seem to have some logic in them though: the percentages seem to be calculated by comparing the new X, Y and Z box dimensions with the original ones before rotating.

See attached model that was orginally placed vertical and was only rotated 90 degrees around its X axis (red).

Hope this can help the team.

[image]

nallath, 2017-07-27:
This has been fixed.

closed 2017-07-27 by nallath

cranking, 2018-12-20:
I think this issue should be reopened -- I'm experiencing the same issue in the UltiCura 3.5