Ultimaker/Cura #774: [2.1.1] 100% infill uses top/bottom speed for infill speed.

rparrett, 2016-06-07:
I'm seeing an issue in 2.1.1 where infill is printed unexpectedly slow. 

In my case, I was slicing at 100% infill, and the issue seemed to be resolved by slicing at 99% infill instead.

It seems that perhaps 100% infill layers in the middle of the model are considered "top/bottom layers" when speed is calculated.

STL:
[14-Corner_Bracket.zip](https://github.com/Ultimaker/Cura/files/302805/14-Corner_Bracket.zip)

GCode @ 100% Infill:
[gk100.zip](https://github.com/Ultimaker/Cura/files/302800/gk100.zip)

GCode @ 99% Infill:
[gk99.zip](https://github.com/Ultimaker/Cura/files/302803/gk99.zip)

Speed Settings:
[image]

Gcode Viewer @ 100% infill:
[image]

Gcode Viewer @ 99% infill:
[image]

Full cura profile:
[infill-speed-profile.zip](https://github.com/Ultimaker/Cura/files/302798/infill-speed-profile.zip)

fieldOfView, 2016-06-07:
> It seems that perhaps 100% infill layers in the middle of the model are considered "top/bottom layers" when speed is calculated.

As far as I understand it, that is exactly what Cura does.

Ghostkeeper, 2016-06-08:
Indeed at 100% infill it sets top/bottom layers to 999999 via inheritance. You can set it back to 8 or so as a workaround, if you like.

At 100% infill, it is more time-efficient to use skin. And it produces better results, since there is no border any more between skin and infill. Of course, one of the downsides is that these are now printed at the speed of skin, which is usually much slower than that of infill. This may defeat the time-efficiency...

closed 2017-11-14 by ChrisTerBeke