Ultimaker/Cura #775: Cura [2.1.2] - Ultimaker 2 No multiple nozzle size

SamGuliver, 2016-06-07:
If you have an Ultimaker 2 and upgraded it with the Olson block, you cannot choose between nozzle sizes in the Software, if you have the right printer selected!
Needs to be fixed!!

fieldOfView, 2016-06-07:
Yeah, we actually had that at one point but unfortunately it got cut... Made me sad to see functional code getting removed.

What you could do is lie, and say you have an Ultimaker 2+. That will get you the nozzle selection, as well as material optimizations.

MarcusWolschon, 2016-07-05:
I can select my nozzle size for an UM2 (not +) extended.
As it doesn't come with an Olsson block as shipped, you may need to enter preferences
and make this setting visible.

bortek, 2016-11-07:
I installed Cura 2.3.0 for my "Ultimaker 2" and I cannot find where to set the nozzle size. If I change to Ultimaker 2+ for example then the option appears on the top where you can select material and nozzle size. But not for "Ultimaker 2". I have enabled all settings in the preferences still nothing. Is this a bug? I cannot use this version of Cura if I cannot set the nozzle size.

fieldOfView, 2016-11-07:
Check the Machine Settings on the Printers pane of the Preferences

bortek, 2016-11-07:
Oh dear it was hidden right there deep under the hood :) 👍  . Any plan to move it out to the surface ?

Ghostkeeper, 2016-11-08:
The Ultimaker 2 only had a 0.4mm nozzle, so others are not supported.

That said, we did have the nozzle size setting exposed as a normal setting for a while. We eventually removed it again because it is technically a machine setting, like gantry height or build volume width, that is not easily modified.

bortek, 2016-11-08:
I think most,  if not all,  um2 users have either olsson block or some other mods to change the nozzle. If you have um2 you would eventually come to the point where you experiment and want ti change the nozzle. That said I  still believe we should have a setting exposed. If not by default then perhaps cia a preference that would make it visible or hidden on the gui just like for um2+

MarcusWolschon, 2017-02-21:
With the new **Cura 2.4**

-  I can not select any nozzle sizes for UM2,
-  can not create a UM2+Olsson Block printer and
-  with the UM2+ selected, I can only select 0.25, 0.4, 0.6 and 0.8mm. However I do all structural parts with a 1.0mm nozzle that can not be selected.

nallath, 2017-02-22:
The UM2+ is with an olsson block by default. I thought it was shipped with a 0.25, 0.4, 0.6 and 0.8

If you want to print with 1mm nozzle, you can also change the line width to 1mm. You will probably need to change a number of other settings due to the different flow to get good results.

MarcusWolschon, 2017-02-28:
The line is not an input field. You can only select between preconfigured nozzle sized.
Doesn't the UM2+ have other differences to the UM2 that may be problematic with this workaround (firmware update, geared down extruder changing speeds, ...)

fieldOfView, 2017-02-28:
*Line Width* is a setting that has an editable number. Changing that from its default value in the profiles you use has ~90% the same effect as changing the nozzle diameter if that were a settable value.

MarcusWolschon, 2017-04-21:
Still exists in Cura 2.5

Ghostkeeper, 2017-04-25:
Again, the Ultimaker 2 has only one nozzle size, 0.4mm. We're not talking about the Ultimaker 2+ here.

Modifications of the UM2 such as the Olsson Block are community-mods. We expect that people able to make a mod such as that to their printer and firmware are also able to modify the JSON file such that the nozzle size is correct, just like people who make other modifications like the size of the head.

Ghostkeeper, 2017-04-25:
> Doesn't the UM2+ have other differences to the UM2 that may be problematic with this workaround

Yeah, well it sorta works, but the extruder is much more powerful so there is less slip going on and this influences the flow very much.

fieldOfView, 2017-04-25:
Correct me if I am wrong, but the Machine Settings dialog lets you set the nozzle size for the UM2, doesn't it?

So you can
* set the nozzle size in Machine Settings, or
* set the line width, or
* set the printer type to UM2+, which gets you 0.25, 0.4, 0.6 and 0.8 mm

There aren't many things in Cura where you have this many options to solve your problem.

MarcusWolschon, 2017-04-26:
"Modifications of the UM2 such as the Olsson Block are community-mods."
Since the Olsson Block update is sold by Utimaker in the shop, I don't consider it a community mod.

nallath, 2017-04-26:
It has also been shipped as part of the UM2 for a while. So I agree with Marcus, it might be a community mod, but it's an "official" community mod.

fieldOfView, 2017-04-26:
Time to revive https://github.com/Ultimaker/Cura/blob/bcf0f507ad63b9a8797563f05965e6960874283f/resources/qml/WizardPages/SelectUpgradedPartsUM2.qml ?

nallath, 2017-04-26:
Yeah, i guess so.

fieldOfView, 2017-04-26:
I'm willing to put time into a PR if you can confirm it is something that will not subsequently get pulled again :/

nallath, 2017-04-26:
It won't be pulled again. Just confirmed this with our product owner.

fieldOfView, 2017-05-02:
Ultimaker has no shop anymore, has it? But the Olsson Block is featured fairly prominently here:
https://ultimaker.com/en/resources/manuals/add-ons
And I am fairly sure the Olsson Block was at one point listed as an addon (in the time they were given away for free with an UM2: https://ultimaker.com/en/blog/17983-free-olsson-block-with-ultimaker-2-and-ultimaker-2-extended)

nallath, 2017-07-27:
You can now select olson block upgrades in machine actions.

closed 2017-07-27 by nallath