Ultimaker/Cura #776: [Cura 2.1.2] Unable to drag files in

RhinoAus, 2016-06-08:
Unable to drag files in from desktop; must use Open dialog box. Very, very slow and inefficient

fieldOfView, 2016-06-08:
Could you see if this also happens if you restart Cura?

What happens on Windows is that the installer must run as "elevated" (admin) user. When it runs Cura after installation, Cura is also ran as elevated user, and that breaks dragging files from the desktop. If you run Cura normally, dropping files should work.

Ghostkeeper, 2016-06-09:
> ran as elevated user, [...] breaks dragging files from the desktop.
:confused:

RhinoAus, 2016-06-09:
Okay; so I ran it normally and dragging works again. Very odd bug :(

fieldOfView, 2016-06-09:
Good to know it is only on the very first run.

fieldOfView, 2016-06-09:
There's a possible (simple) fix which we will have to include in our installer:
http://stackoverflow.com/questions/16555625/how-to-launch-an-executable-on-the-end-of-installation-correctly

closed 2016-06-09 by fieldOfView