Ultimaker/Cura #778: Cura [2.1.2] -profile cannot save  bug???

QoooQ, 2016-06-08:
i change value and update but can not save?  Why?

when i change valuse then select other profile    it ask me want update profile  i select update  but value is not save ?! bug???

Ghostkeeper, 2016-06-09:
Sounds like a bug to me. If you say you want to update the profile it should update the profile you were coming from before switching to the new profile.

QoooQ, 2016-06-09:
please try   it      and you will find bug(no save)!!

carbogit, 2016-06-17:
Confirm, some settings, esp. the material section, do not save. Retraction distance stays 6.5 mm whatever I do.

Ghostkeeper, 2016-06-17:
We've since fixed this issue per https://github.com/Ultimaker/Uranium/pull/133

The issue was that material profiles were overriding the user profiles. User profiles are now always on top.

closed 2016-06-17 by Ghostkeeper

DarkRabbit19, 2016-07-11:
I stil have te problem

fieldOfView, 2016-07-11:
@DarkRabbit19: with what version? 
This has been fixed in Cura 2.2, which is not yet released.