Ultimaker/Cura #780: Edit start / end gcode

raoul596, 2016-06-09:
I have downloaded Cura 2.1. I want to edit the start gcode so I can enable auto bed leveling (G29). Where can I do that?

fieldOfView, 2016-06-09:
Unfortunately we don't currently have a way to edit the start/end gcode from within Cura. You can edit it in the printer definition file. We know this is not ideal, and will reintroduce the functionality in a later version.

Ghostkeeper, 2016-06-09:
That would be in `<Cura's installation folder>/resources/machines/<printer>.json`.

For example, `C:\Program Files\cura\resources\machines\ultimaker2.json`.

thester198, 2016-06-21:
Alright so what if I want to do is add the M80 (ATX Power) command right after the bed temp is reached but before the hotend temp is set . as it seems to be sent before the machine json setup is read. It could be sent first too but after the bed would be ideal. Any way for me to modify the code there?

dteeter, 2016-06-21:
How would you manually edit the start/end gcode on a Mac? 

When will this essential feature (start/end gcode editing) become available in 2.1.X?

fieldOfView, 2016-06-21:
Cura 2.1 is feature-frozen; no features are going to be added to it.
Editing machine parameters is likely going to be added in 2.2 or 2.3.

Ghostkeeper, 2016-06-21:
> Alright so what if I want to do is add the M80 (ATX Power) command right after the bed temp is reached but before the hotend temp is set . as it seems to be sent before the machine json setup is [...]

If there is a M104 or M109 command in the start g-code, put your M80 command right before it. If there isn't but there is an M140 or M190 command in the start g-code, put your M80 command right after it. If there is neither, you could try just adding M80 to the end of the start g-code, but then I think you'll need to write a post-processing script instead to insert M80 commands before the nozzle temperature commands (M104/M109). If there's no M104, M109, M140 or M190 commands in the entire resulting g-code, then you're out of luck: The printer heats up the bed and nozzles before starting even reading the g-code, so there's nothing you or Cura can do.

CCS86, 2016-07-01:
What is the syntax for changing the start gcode in the machine cfg file, instead of the json? I'm finding that it works, but is inserting **exactly** what I write there, and not acknowledging that "\n" represents a new line:

[general]
name = Ultimaker 3-4
type = ultimaker_original
active_profile = Solid with support
version = 1
material = PLA

[machine_settings]
machine_heated_bed = True
machine_extruder_drive_upgrade = True
machine_start_gcode = {G21 ;metric values\nG90 ;absolute positioning\nM82 ;set extruder to absolute mode\n....

fieldOfView, 2016-07-02:
I am not 100% certain that setting machine_start_gcode in the machine_instance is supported (for the exact reason you're finding out). You can edit the value in the machine json file instead, but that means it is shared with all instances of that printer.

CCS86, 2016-07-02:
Right, I was hoping to leave the JSONs as master files, so to speak, and just manipulate my instances. You may be right, that there is no correct syntax because it hasn't been fully implemented. 

I'm still hoping someone will chime in with the formatting to make it work!

fieldOfView, 2016-07-02:
> Right, I was hoping to leave the JSONs as master files
And that is very sound reasoning.

Lots of work has been done between the feature-freeze of 2.1 and what is now master. This should make it easier to (re)add editing start- and end-gcode and editing machine settings in general further down the line, though work has not yet started on either of those because we're working on other stuff.

Ghostkeeper, 2016-07-03:
What happens if you format it like the shaders?
https://github.com/Ultimaker/Cura/blob/master/resources/shaders/overhang.shader

CCS86, 2016-07-03:
I tried formatting like this, but no dice:

u_ambientColor = [0.3, 0.3, 0.3, 1.0]

fieldOfView, 2016-07-03:
See this thread on our forum:
https://ultimaker.com/en/community/20538-cura-21-open-beta-has-been-released?page=10#reply-150273
I think there needs to be whitespace at the start of each line. Or something.

CCS86, 2016-07-03:
@fieldOfView Thank you, that works perfectly!

closed 2016-07-04 by nallath

jrisher1990, 2017-07-04:
I am trying to figure out where to even go to get to the screen to edit the gcode because I have yet to be able to find it. Can someone send me in the right direction because my Anet A6 does not auto home after the print is complete it just shuts down right on top of the print and if I am not there to make it auto home it melts a spot on the top of the print where it sits so I would like to edit the code to make it automatically go home when done but I would also like to have the bed come forward instead of go all the way back so I can get the print off the bed. Someone once told me what codes to add to make it perform those but never really told me how to get to the edit screen. When I go to extensions and edit gcode it only has select codes already preset and I can figure out how to get around that.

fieldOfView, 2017-07-04:
In the sidebar, click on the name of your printer. From the dropdown, choose "Manage Printers...". In the window that opens, press "Machine Settings".

jrisher1990, 2017-07-06:
I did all of that and I see the start gcode and end gcode at the bottom but it wont let me add or do anything to it.

fieldOfView, 2017-07-06:
Are you saying that when you click in the start/end gcode fields, you cannot enter any text? That would be very strange. (and something I have not seen reported before). Can you share a screenshot?