Ultimaker/Cura #784: WARNING: Application calling GLX 1.3 function "glXCreatePbuffer" when GLX 1.3 is not supported!  This is an application bug!

alexpacini, 2016-06-10:
I am using the software remotely with X2Go and if I launch the application it will stuck into the splash screen. I thus tried launching Cura from the shell and I have got this message: "WARNING: Application calling GLX 1.3 function "glXCreatePbuffer" when GLX 1.3 is not supported!  This is an application bug!
".
I tried the open beta and there it was not working, but it was on the closed beta.
I can say that X2Go does not support GLX 1.3, but because the closed beta was working without GLX1.3, is it necessary here?

Thanks, Alex

Ghostkeeper, 2016-06-10:
Seeing that anywhere on other OpenGL applications? It smells like something between X2Go and GLX.

alexpacini, 2016-06-10:
In some others may happen. For example KDE5 does not work on X2Go, but XFCE
works well. glxgears works, for example.
Alex

---
Alex Pacini, MS
email: alexpacini90@gmail.com
blog: https://pacinispace.blogspot.com
ORCID: orcid.org/0000-0002-0316-4989
skype: alex_pacini

---

awhiemstra, 2016-06-13:
We require OpenGL 2 with Framebuffer Objects for the new Cura. I am not sure what the current level of support for that is in X2Go. The createPBuffer call likely comes from within Qt when it tries to query the display properties. Have you tried any other Qt5 applications? Since you mention KDE5, I am guessing those also run into issues?

alexpacini, 2016-06-13:
Yes, they have issues. It is only strange that the closed beta worked but
the open beta and the now official one does not, all of them being the Qt5
new version of Cura.

---
Alex Pacini, MS
email: alexpacini90@gmail.com
blog: https://pacinispace.blogspot.com
ORCID: orcid.org/0000-0002-0316-4989
skype: alex_pacini

---

nallath, 2017-07-27:
I'm going to assume that this has been solved already. If not, feel free to re-open or comment on it again.

closed 2017-07-27 by nallath