Ultimaker/Cura #789: Extruder motor not working

glostick, 2016-06-10:
I have a Prusa I3 first off. So I have been using Cura 15.04.2 for almost a year with no problems. I found Cura 2.1.1 for windows 64-bit and used that with no problems. Now I installed Cura 2.1.2 for windows 64-bit and my extruder motor will not work. The extruder motor works with Cura 15.04.2, which is still installed on my PC. Cura 2.1.1 seems to print out a much better print then Cura 15.04.2.

If anyone can help me with this I was love the feed back. I will try to uninstall Cura 2.1.2 and re-install Cura 2.1.1, since I saved the exe for a just in case. I will let you guys know if that works or not.

thopiekar, 2016-06-10:
If possible slice with (at least almost) the same settings with Cura 2.1.1 and 2.1.2. and send us here the GCode. We can compare and maybe find the reason.

glostick, 2016-06-10:
So my printer will not print with 2.1.1 now. I think once I installed 2.1.2 the profile just crapped out. So I used the default profile and it still did not work. I will upload the gcode from 15.04.2, 2.1.1, and 2.1.2

[212Norm_Marvin.txt](https://github.com/Ultimaker/Cura/files/309951/212Norm_Marvin.txt)
[211Marvin.txt](https://github.com/Ultimaker/Cura/files/309952/211Marvin.txt)
[15024Marvin.txt](https://github.com/Ultimaker/Cura/files/309953/15024Marvin.txt)

BerndJM, 2016-06-12:
In your 15024-gcode you have a hotend temp at 210 and in bot 2xx-gcodes ist' only 175c
so I think a firmware "cold extrusion prevention" kicks in

glostick, 2016-06-12:
in 15024-gcode I was using wood filament and in 2xx-gcode, I just bought a new PLA filament that has a temp of 175 to 200c. So no matter which one I use the motor did not move, plus I even raised the temp to 255 thinking there was a clog, which there was not.

Ghostkeeper, 2016-06-12:
The only change we did for 2.1.2 was to fix the change log. It was displaying a white page.

I'm thinking something went wrong with the driver installation during the set-up. Perhaps you should run through the set-up again?

glostick, 2016-06-12:
Ok, I will give that a try and let you guys know. Thank you everyone for your help.

glostick, 2016-06-13:
Ok, guys I think I fixed it. So I removed Cura 15.04.2 and 2.1.2 and reinstalled it, no dice. So just for the hell of it I downgraded the firmware and then I upgraded the firmware again, that seemed to fix the problem so far. Thank you to everyone.

Ghostkeeper, 2016-06-14:
I'm going to file chalk this up as a glitch in Arduino's driver installer then.

closed 2016-06-14 by Ghostkeeper