Ultimaker/Cura #793: Cura 2.1.2 - Raft Height Not Used in Size Calculation

ZhuDaHai, 2016-06-12:
When Cura determines size of object is within bounding area, the height of the Raft (when used) is not taken into account.
1. Scale object to max height of build area
2. Enable Raft (no warnings and object stays yellow (not gray))
3. Slice and Save
4. Note that top layers of object are outside max build height.
i.e: My printer max height is 180mm. The printer firmware will not allow head above this point).
But gcode shows command to set Z location above 180mm:
`;LAYER:899
G1 F2400.000 E1057.03538
G0 F7200.00000 X89.713 Y75.833 Z181.460`

Ghostkeeper, 2016-06-12:
Good point. We'll get a fix scheduled.

ianpaschal, 2018-02-21:
No more work will be done on versions 2.x. If this is still occurring in versions 3.2 or newer, we can re-open this issue.

closed 2018-02-21 by ianpaschal

Ghostkeeper, 2018-03-13:
This was fixed quite a long time ago.