Ultimaker/Cura #794: Cura stops at "preparing to slice..."

worker72, 2016-06-12:
Yesterday downloaded and installed Cura 2.1.2. Sliced an object, and printed it fine.
Today, I want to slice a similar object (my next iteration), but Cura hangs at "preparing to slice".

Cura 2.1.2
Windows 10

Engine log:
Calling engine with: ['C:\\Program Files\\Cura 2.1\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Cura 2.1\\resources\\machines\\ultimaker_original_plus.json', '-vv']

worker72, 2016-06-12:
After restarting Cura, slicing worked ok.

Ghostkeeper, 2016-06-12:
We've had reports where this only happens the first time after installation or after deleting your settings.

PrinterMan, 2016-06-16:
I noticed that the print bed size has shrunk with the last 2 versions and this is the reason that its hanging on preparing to slice. What is up with the print size shrinking?

fieldOfView, 2016-06-16:
You mean in height? Check the print-sequence setting, it it probably set to "One at a time". Try setting it to "All at once". 
The height is limited in "One at a time" mode because otherwise the printhead or gantry could collide with a previous object within the job.

luukakkerman, 2016-06-17:
I have encountered the hanging at 'preparing to slice' now multiple times over the last weeks with Cura 2.1.0-
Indeed, restarting the program works to get at least the model slicing when importing again. Sometimes moving, scaling, changing print settings and doing all action in general less quickly looks to solve this. But some models (and no, they are not complicated at all but very basic geometrical models) seem to activate this hang bug every time.

As I see 2.1.2 is out, I will try this release now to see if it helps.

nickthetait, 2016-06-17:
If you run into a model that is able to consistently trigger this bug please post it here.

johnwebbcole, 2016-06-23:
The peg and collar from http://www.thingiverse.com/thing:716747 will both hit this when scaling is turned off, or if its on (and it always resizes these) and you remove the object transformations.  1.5.4 seems ok with both of them.

I'm on 2.1.2 on macOS 10.11.5

universam1, 2016-07-14:
What does help for me is to run "Center Object on Platform" for to get it starting.
Hope that helps

Ghostkeeper, 2016-07-15:
Basically the workaround is to trigger a reslice then.

We've seen this issue come up from time to time, even currently on 2.1.99.

blacktea, 2016-07-20:
For me slicing stops when I have more objects (like 5 or more) and when I'm changing "PRINT SEQUENCE" parameter a lot of the times the slicing does not react and despite me changing all the other parameters slicer does not "reslice" the objects.

Cura 2.1.2 64bit
Windows 10

cottmann, 2016-07-20:
Same here. Cura 2.1.2 on Windows 7 and Ubuntu 14.04 hangs at "preparing to slice...". 
Using CuraEngine via the command line works (Ubuntu).
Smaller models work fine. 

This one worked (logs from Windows version): 
```
2016-07-20 20:55:11,843 - DEBUG - Calculating normals took 0.0 seconds
2016-07-20 20:55:11,844 - DEBUG - Loaded a mesh with 8160 vertices
2016-07-20 20:55:11,860 - DEBUG - Loading mesh took 0.5820291042327881 seconds
2016-07-20 20:55:12,621 - DEBUG - Attempting to kill the engine process
2016-07-20 20:55:12,621 - DEBUG - Killing engine process
2016-07-20 20:55:12,621 - DEBUG - Engine process is killed. Recieved return code 1
2016-07-20 20:55:12,792 - INFO - Backend crashed or closed. Restarting...
2016-07-20 20:55:12,901 - DEBUG - Attempting to kill the engine process
2016-07-20 20:55:12,901 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-07-20 20:55:12,933 - DEBUG - Backend connected on port 49674
2016-07-20 20:55:22,109 - DEBUG - Autosaving preferences, instances and profiles
2016-07-20 20:55:26,536 - DEBUG - Backend quit with return code 0. Resetting process and socket.
2016-07-20 20:55:26,536 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-07-20 20:55:26,551 - DEBUG - Backend connected on port 49674

```
And here it didn't: 
```
2016-07-20 20:59:20,242 - DEBUG - Calculating normals took 0.0009999275207519531 seconds
2016-07-20 20:59:20,243 - DEBUG - Loaded a mesh with 8982 vertices
2016-07-20 20:59:20,262 - DEBUG - Loading mesh took 0.39801788330078125 seconds
2016-07-20 20:59:20,989 - DEBUG - Attempting to kill the engine process
2016-07-20 20:59:20,989 - DEBUG - Killing engine process
2016-07-20 20:59:20,989 - DEBUG - Engine process is killed. Recieved return code 1
2016-07-20 20:59:20,989 - INFO - Backend crashed or closed. Restarting...
2016-07-20 20:59:21,098 - DEBUG - Attempting to kill the engine process
2016-07-20 20:59:21,098 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-07-20 20:59:21,129 - DEBUG - Backend connected on port 49674
2016-07-20 20:59:29,047 - DEBUG - Attempting to kill the engine process
2016-07-20 20:59:29,047 - DEBUG - Killing engine process
2016-07-20 20:59:29,047 - DEBUG - Engine process is killed. Recieved return code 1
2016-07-20 20:59:29,281 - INFO - Backend crashed or closed. Restarting...
2016-07-20 20:59:29,385 - DEBUG - Attempting to kill the engine process
2016-07-20 20:59:29,403 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-07-20 20:59:29,451 - DEBUG - Backend connected on port 49674
2016-07-20 20:59:30,481 - DEBUG - Autosaving preferences, instances and profiles
```

Ghostkeeper, 2016-07-21:
Yes, the margin is the width of the brim. You can reduce the brim width to reduce that margin (but your brim will be smaller).

Ghostkeeper, 2016-07-21:
In 2.1.2 when it can't fit the models on the plate due to the disallowed areas of one-at-a-time printing, it gets stuck on Preparing to Slice... That one is known to us. It seems different from what worker72, PrinterMan and luukakkerman had though.

cottmann, 2016-07-21:
Thank you for clarifying, @Ghostkeeper. I clearly overlooked the brim. 
And thank you for a great piece of software!

awhiemstra, 2016-07-21:
> In 2.1.2 when it can't fit the models on the plate due to the disallowed areas of one-at-a-time printing, it gets stuck on Preparing to Slice... That one is known to us. It seems different from what worker72, PrinterMan and luukakkerman had though.

Also, in 2.2 it will provide a clearer indication of this behaviour.

thopiekar, 2016-09-18:
I guess it has been fixed now..

closed 2016-09-18 by thopiekar