Ultimaker/Cura #798: List of all variables for JSON file

boelle, 2016-06-12:
Hi

I'm building a JSON for the mendel90 printer but i lack information on ALL the variables that can be used and with explanation on what each variable do

fdmprinter.json gives some clue but is that the complete list or are there other variables ??

Ghostkeeper, 2016-06-12:
You were close.

There are two files now which contain ALL settings:
* [fdmprinter.def.json](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json)
* [fdmextruder.def.json](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmextruder.def.json)

It is in theory possible for a printer definition to define additional settings (we used to have that kind of construction for Ultimaker Original to track whether a printer had the extruder driver upgrade kit) but that is now not used by any printer.

ZhuDaHai, 2016-06-13:
Not all variables are implemented, right?
acceleration_print, while defined, doesn't seem to impact print time estimates.

Ghostkeeper, 2016-06-13:
All leaf-settings of the setting tree (that have no children) are directly used by the engine, with the exception of a few machine settings such as machine_width that are only used by the front-end. Parent-settings are all used in some way via inheritance to change the values of their children.

I thought that the time estimate was accurate enough to actually include the acceleration and jerk in the estimate. I seem to remember that passing by recently in a code review. Maybe the effect is just too small. @BagelOrb would know.

boelle, 2016-06-13:
this is great...

just knowing that you have the complete list of settings removes a lot of the confusion factor

fdmprinter.def.json.... i noted that some of the lines are just cut off... this is bad as some of the explanation/comments are lost

Ghostkeeper, 2016-06-13:
> fdmprinter.def.json.... i noted that some of the lines are just cut off... this is bad as some of the explanation/comments are lost

I think that's just a display issue in your web browser. Middle-click the text to scroll right :wink:

boelle, 2016-06-13:
correct :-D

and i learned something new

closed 2016-06-13 by Ghostkeeper

boelle, 2016-06-13:
a q even thou this one is closed:

1) what are the requirements for getting a machine included with cura?
2) is it possible to add material files that are spec to a filament vendor/seller ?

boelle, 2016-06-13:
also i looked in the file for the ultimaker original and see

"has_materials": true,

but that one is not in either of the 2 files above

boelle, 2016-06-13:
http://pastebin.com/sUYEXyVz

how does that look?

dont mind start and end gcode... i need to remove temp stuff from there

machine_head_shape... those i cant find documentation on either

fieldOfView, 2016-06-13:
Your pastebin version is a bit of a mixup between the 2.1 format and the format we use now in master (sorry for the confusion). A machine profile you would like to use in 2.1.2 (the current version) has to use the "old" format we use in the 2.1 branch. You can find the relevant files for 2.1 here:
https://github.com/Ultimaker/Cura/tree/2.1/resources/machines

The following is only relevant if you want to make a machine defintion for master
"has_materials" is in the metadata section. The metadata section is not described in fdmprinter.
"pages" is currently not implemented in the master branch, but should soon return one way or another

boelle, 2016-06-13:
it would have been easy if i could just export settings from 15.04.6 :-D

on the other hand if i get this right i will have far better understanding of the settings

will try and get this done and then make a new pastebin.

awhiemstra, 2016-06-13:
> 1) what are the requirements for getting a machine included with cura?

Generally, you need a working definition file with a platform mesh and optionally a texture.

> 2) is it possible to add material files that are spec to a filament vendor/seller ?

Yes, though this bit is currently still in progress. We are working on a solution to make this easier even.

> also i looked in the file for the ultimaker original and see
> "has_materials": true,

This is metadata, which is currently not yet fully documented. The four metadata properties "has_variants", "has_materials", "has_machine_materials" and "has_machine_qualities" indicate how Cura should deal with variants, materials and qualities:

- has_variants indicates whether there are variant profiles for the machine. Variant profiles provide overrides of machine settings like nozzle size. Currently only the UM2+ is using them to handle different nozzle sizes.
- has_materials indicates whether the machine supports selecting different materials from within Cura. Most notable here is the UM2 that handles all material related things on the machine.
- has_machine_materials indicates whether Cura should filter the list of materials by the machine - specifically, it means there are machine-specific material profiles available.
- has_machine_qualities does the same as has_machine_materials but then for quality profiles.

BagelOrb, 2016-06-13:
_machine_width_ **is** actually used in the engine. Other settings like _machine_head_shape_ are used by the frontend only. That settings is used to determine the distance objects are supposed to be kept apart in one-at-a-time mode. It describes the 2d polygon of the area in which the print head might knock over other printed parts; so it's the shadow of your print head projected down.

I don't know why there is a _machine_head_with_fans_polygon_ and a separate _machine_head_polygon_ . @nallath do you know?

BagelOrb, 2016-06-13:
Note that all machine settings are described in the json files on the master branch ;)

boelle, 2016-06-13:
hehe... i reverted to just having the basic settings...

not confused at all there :-P

boelle, 2016-06-13:
side note... also once tried to load the um2.1 firmware on my printer but had to give up

that was only to get as close to UM as i could on a reprap

boelle, 2016-06-13:
for my platform stl i simply opened the openscad files that is used to build the mendel90 and removed everything except the bed and glass.... 

i of course have the option to export the whole machine as an stl... would that confuse cura to have an stl like that? or would i simply have to adjust the offset so that the print volume is clear of the stl?

its just for showoff effect

boelle, 2016-06-13:
retraction_combing

in 15.05.6 i have this set to all... 

i try to put this in the "overrides":  section in the JSON

but i cant find any example use of it

awhiemstra, 2016-06-13:
It would probably work but block a lot of the view to have the entire machine. So reducing it to something simple is probably a good idea. The main purpose of the platform is to easily distinguish different machine types.

boelle, 2016-06-13:
oki

i decided to manual copy over my settings from 15.04.6

if i export settings as a cura profile could those be integrated in to the machine JSON or how does it work?

boelle, 2016-06-13:
my final JSON:

http://pastebin.com/QZjFjFPQ

cura profile:

http://pastebin.com/Z9nmKjnD

next "problem"

how do i print via usb?

nallath, 2016-06-13:
I don't know why there is a machine_head_with_fans_polygon and a separate machine_head_polygon . @nallath do you know? -> Per object setting magic. It's not easy to explain though.

boelle, 2016-06-13:
the only thing i can think of is for machines that have a cold end fan and for those that do not

in my own case i will just set gantry height to distance from nozzle tip to underside of cold end fan as that is the first thing that comes

nallath, 2016-06-13:
The gantry height, in the case of the Ultimaker is the distance from the nozzle to the head y-axis rod.

boelle, 2016-06-13:
yep... i'm trying to make a JSON that can go with cura on next release

its for the mendel90 type of printer and i want it to be as perfect as can be with nothing left out... but on gantry height yes i'm cheating :-D

boelle, 2016-06-13:
Q... has printing over USB been removed?

nallath, 2016-06-13:
Nope. It's still there.

fieldOfView, 2016-06-13:
@boelle: are you talking about master or 2.1.2?
Master is currently "in motion", and USB printing may not yet be posible.

boelle, 2016-06-13:
not sure... its the 64 bits one you can download from here

https://ultimaker.com/en/products/cura-software

fieldOfView, 2016-06-13:
That page says "2.1.2". 
USB printing should be available.

boelle, 2016-06-13:
i only have save to file

http://imgur.com/xqbDQcA

nallath, 2016-06-13:
That means that Cura was unable to detect your printer.

fieldOfView, 2016-06-13:
Is there no option to print over USB if you press the little downward arrow on the right of the "Save to File" button?

PS: note that we also have a forum (https://ultimaker.com/en/community/software-and-firmware/cura-and-marlin?topics=1&sort=latest&type=all). All of our developers get email notifications about all of your short posts.

boelle, 2016-06-13:
nope...and i use repetier as firmware

boelle, 2016-06-14:
just to make it clear... setting them by hand does make a difference

> works fine under 15.04.06... but only when i tell what com port and speed. auto detect never worked

BagelOrb, 2016-06-28:
Any update on this?

boelle, 2017-04-08:
i'm still on 15.04.06 as its the only thing i can get to work

as soon  i set my comport and speed it connects. Cura 2.4 still does not auto detect

i just wish there where some file somewhere i could edit in notepad to tell where my printer is

BagelOrb, 2017-04-08:
Ultimaker printers weren't really made for USB printing, so Ultimaker isn't
really going to invest time in this.

You are welcome to edit the code and get it to work. If you do get it to
work, you could share your code so that every USB connected printer also
benefits.

Otherwise you could always put the gcode on a memory stick and transfer it
to your printer by hand.

boelle, 2017-04-09:
i dont have any clue on doing code, but hardware is another thing

@fieldOfView better look through the PR's first. (i have not but will)

could be awesome to have some kind of control of what port etc it uses... does not have to be in the interface but could be as little as a JSON setting that overrides the automatic

fieldOfView, 2017-04-09:
Well, there's this: https://github.com/Ultimaker/Cura/issues/1049

stuartpb, 2017-12-29:
I'm unclear on the functional definition of `has_materials`. Would setting this to `false` mean that the printer is only capable of printing one kind of material, and should not allow material selection in Cura? Are there printers like that?

fieldOfView, 2017-12-30:
Setting ```has_materials``` to false causes the material selection to go away in the UI, so there won't be settings that change per material (because there are no materials to choose from). A user can still create profiles as normal, and name them after materials.