Ultimaker/Cura #799: 2.1.2 cant detect my printer

boelle, 2016-06-14:
But in 15.04.6 it cant detect it automatic either

But if select the right com port and speed it works like a charm

Why was the manual selection removed?

fieldOfView, 2016-06-14:
Technically it was not removed. It just was not added to Cura 2.1. Yet.

Cura was rewritten from scratch between Cura 15.04.x and Cura 2.x. Selecting a com-port and speed has not been added back in.

boelle, 2016-06-14:
will it be added back at some point?

Ghostkeeper, 2016-06-15:
Going to take a while, but manual comport-selection would be useful to add.

boelle, 2016-08-08:
when is this going to happen?

Ghostkeeper, 2016-08-09:
It's scheduled for release 2.3.
2.3 would be due a few months after 2.2, which is scheduled for final release in 6 weeks. So somewhere towards the end of the year, probably. We don't rightly know.

boelle, 2016-08-09:
that at least gives some clue to the timeframe

boelle, 2016-09-14:
just tried the beta of 2.3... com port settings is still the big showstopper

i assume there is still a chance that it will be added

boelle, 2016-09-14:
just curious... how does cure do this automatic? it must do some scanning and what if 2 printers are connected how do you get the right one?

Ghostkeeper, 2016-09-15:
It seems that 6 days ago this issue was moved to release 3.1.

Ghostkeeper, 2016-09-15:
Cura tries to connect to every active COM port and sends a g-code to it at various baud rates that requests the temperature. If Cura receives a string back that represents an integer, it assumes that that was the temperature and that there is a printer behind the COM port.

It just takes the first COM port that gives a valid temperature, I believe.

boelle, 2016-09-15:
ahh.. so being able to select yourself still makes a bit sense :-D

but will wait for 3.1 and see how it works

maybe add a 3.1 label to this one?

boelle, 2016-09-15:
btw... i run repetier firmware on my printer.. not sure if they are compatible with the automatics and i just have to change a setting

also have a K8200 that i plan to install marlin on... maybe repetier here also

Ghostkeeper, 2016-09-15:
Actually, it seems that it continues trying after a printer is found, and adds all printers (and a Print to USB option for each). It doesn't stop after one.

If it answers with any integer within 5 seconds of receiving "M105", it is accepted as a printer. Marlin does this. Repetier too, according to [this](http://reprap.org/wiki/G-code#M105:_Get_Extruder_Temperature).

boelle, 2016-09-15:
strange as repetier does

>>> M105
SENDING:M105

T:25.47 /0 B:25.12 /0 B@:0 @:0

boelle, 2016-09-15:
this is done from pronterface

fieldOfView, 2016-09-15:
This issue (https://github.com/Ultimaker/Cura/issues/736), though a long read, has some things you could try. Anyhow, this issue is a duplicate.

closed 2016-09-15 by fieldOfView