Ultimaker/Cura #802: Retraction Distance setting reseting

chinzw, 2016-06-15:
When i change the retraction distance from the default of 6.5 it always returns to that value, saving a new profile, or updating doesn't work.

chinzw, 2016-06-16:
I think i found the culprit of this.
the Material settings are overriding the Profile settings and by default the material profiles have this line in them:
retraction_amount = 6.5

fieldOfView, 2016-06-16:
Good find. If you remove that line from the material profiles, it should indeed work. All of this has been fully reworked for 2.2 which will handle this much better.

closed 2016-06-16 by nallath

soldanr, 2016-06-16:
Thanks. I just verified that!
Opened resouces\profiles\materials\pla.cfg and changed the settings.

[general]
version = 1
type = material
name = PLA

[settings]
material_bed_temperature = 55
retraction_amount = 1.5