Ultimaker/Cura #807: 2.1.2 {} Replacements Not Working

chinzw, 2016-06-16:
Hi guys, these replacements don't work anymore.
For example this doesn't get the proper replacements like it used in cura 15:
"machine_start_gcode": {
            "default": "M190 S{print_bed_temperature}\nM104 S{print_temperature}\nG21\nG90\nM82\nG28\nG29\nG1 Z0.0 F9000\nM109 S{print_temperature}\nG92 E0\nG1 F9000\nM117 Printing..."
}

chinzw, 2016-06-16:
Never mind, replacements are working, but variable names have changed.
material_print_temperature
material_bed_temperature

thopiekar, 2016-06-16:
Which printer do you use?
Wouldn't you like to correct that here? 😉

chinzw, 2016-06-16:
It is working as intended, the problem i had is that i copied my start gcode from cura 15, but the variables had a change in names.

closed 2016-06-16 by chinzw

thopiekar, 2016-06-16:
Ok then 👍