Ultimaker/Cura #808: Add Mendel90 to cura 2.1.2

boelle, 2016-06-16:
attached is the JSON for the Mendel90

NOTE: rename it from mendel90.PNG to mendel90.JSON

[image]

Ghostkeeper, 2016-06-16:
Added the files. Thanks for your contribution! The JSON required some structural modifications, but the values are still the same. I'm guessing that you meant for the machine head shape the minimum-x and minimum-y to be negative, otherwise you have a collision rectangle that doesn't even enclose the nozzle itself.

closed 2016-06-16 by Ghostkeeper

boelle, 2016-06-16:
dooh.. yes

and also not sure if i got all variables in there that describe the hardware

there should be a generator for this :-D or some better documentation for dummies :-P

Ghostkeeper, 2016-06-16:
There was a generator, but it's outdated. Eventually we'd like to allow people to make new machines "in-game".

boelle, 2016-06-16:
that would be a good thing... 

i tried to test the JSON... but since i cant select comport and speed i had to stop where i'm now... once the comport selection is added i can test and adjust it.

mostly the start and end Gcode needs some tweaks