Ultimaker/Cura #809: Print Bed Size Smaller; Cant print like before?

PrinterMan, 2016-06-16:
I noticed that the print bed size has shrunk with the last 2 versions and this is the reason that its hanging on preparing to slice. What is up with the print size shrinking?

fieldOfView, 2016-06-16:
You mean in height? Check the print-sequence setting, it it probably set to "One at a time". Try setting it to "All at once".

The height is limited in "One at a time" mode because otherwise the printhead or gantry could collide with a previous object within the job.

PrinterMan, 2016-06-16:
No in width a 209mm object won't print now.

PrinterMan, 2016-06-16:
In older versions of cura we could utilize more of the bed space, it was 225mm limit. I know there was an issue with the right fan shroud hitting the wall though to shrink down the size of the bed isn't the cure. What needs to be done is utilize all of the space that can be printed on the glass and readjust the center to the left more to make use of the space that is lost.

Or Ultimaker could provide new, slimmer fans and then we can actually use the 225mm build volume as advertised.

Ghostkeeper, 2016-06-17:
That's strange, because it says the limit is 223mm: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/ultimaker2.def.json#L24

PrinterMan, 2016-06-17:
Open Cura 2.1.2 the latest build software for Mac and try uploading a model that is 209mm long, you wont be able to print it. It's been getting smaller and smaller but Ill tell you right now that theres no way to print 223mm therefore that limit is false advertising.

nickthetait, 2016-06-17:
Would you be able to provide a link to an example model that does this? Additionally adding a screenshot might help figure out what is going wrong?

PrinterMan, 2016-06-18:
I went ahead and deleted the machine profile and re-added the Ultimaker 2 Extended and this seemed to fix the error. I believe there should be a setting which allows you to make adjustments to the printer profile.

Also I noticed that for the Ultimaker 2 Extended+ there is an option for "Material" and "Nozzle" yet the Ultimaker 2 Extended doesn't have those settings. I believe these settings should also be for the non + models.

nallath, 2016-06-20:
Changes to printer profile; You can do this. Change the json files.
As for the nozzle profiles; We had those, but were asked to remove this. 
Spaces; Because we use URL-encode.

Ghostkeeper, 2016-06-20:
Maybe you still had the machine instance leftover from back when we had a bug that swapped the y and x axes.

PrinterMan, 2016-07-02:
Why do we have to change the json files when before we could make changes to the printer in the menus? This is a feature that should have been kept as I don't have anything that can open a json file and can't even locate them, all I have are cfg files.

The point of software is to make it user friendly and this issue just takes away bed space, at the moment Im having the same issue again after reinstalling Cura.

fieldOfView, 2016-07-02:
> Why do we have to change the json files when before we could make changes to the printer in the menus?

Because Cura was rewritten from scratch. It is not a question of having thrown the functionality out, but more of not having had time to implement it again in the new version.

PrinterMan, 2016-07-02:
Is this a feature that could be added? One fix would be to correct the parameters of the build plate of the Ultimaker's by using the whole build plate with:

1. A few mm shaved off the right side to prevent the fan from hitting the wall, and only as much as necessary.
2. Instead of a square shaped bed that is cut off at the tabs use the whole build plate volume as advertised though make necessary adjustments:

A. Re-center the center of the machine with respect to the few mm shaved off the right for fan shroud to wall protection.
B. Create an error/grey out of bounds field ONLY for the tabs and not for the useable glass between the tabs.

Ghostkeeper, 2016-07-03:
Regarding A, we did tune the build plate volume as far as it could go, though there is naturally some variation, especially in the Z-dimension due to the screws under the build plate. Then we were called back by marketing that the build plate volume should match the build plate volume as advertised on Ultimaker's website.

Regarding B, the disallowed areas *are* only for the clips, not for the usable glass between the clips. However there is also a bit of grey due to the brim, raft or skirt. It was easier to expand disallowed area around the build plate than to expand the collision box of the mesh, because with brim/skirt/raft the meshes can still be placed closer to each other than the brim/skirt/raft width. Only not closer to the edge of the build plate.

Modifying machines from the GUI will come back, but we just hadn't the time yet.

ianpaschal, 2018-02-26:
This has not been discussed in a year and a half, so I'm closing it as a non-, or no-longer-relevant issue.

closed 2018-02-26 by ianpaschal