Ultimaker/Cura #811: Printer Axis Origin (Cartesian )

soldanr, 2016-06-17:
It would be great to see the Cartesian origin in the bounding box of the printer area again to give a frame of reference on the bed like in the old Cura. It makes it a lot easier for teaching purposes to people unfamiliar with motion controllers.

[image]

Ghostkeeper, 2016-06-20:
Agreed. What do you think would look nice for the case when the axis origin is in the centre of the bed though?

soldanr, 2016-06-20:
I like to teach the right hand-rule for the axis layout. I think if we had an X-axis and a Y-axis it will be easy to visualize the Z-axis. It would be neat to have a semi-transparent X and Y axis only so we don't bother with a Z axis on the center of the plate. This way we know at last which quadrant is positive and negative on both axis. I like the bluish arrows to match the color of the GUI. Here are some options:

[image]

fieldOfView, 2016-06-21:
I like the subtle marks on the middle of the buildplate edges. The red/green axis indication on the buildplate origin also works with the overall feel of the app.

The mark in the center of the buildplate looks a bit out of place to me. I think a "center-indication" (it's not the origin!) could be useful though.

Ghostkeeper, 2016-06-21:
> (it's not the origin!)

It could be. When `machine_center_is_zero` is true, it is the origin.

fieldOfView, 2016-06-21:
It's... complicated ;-). The translate tool presents it as the origin, regardless of the machine_center_is_zero setting.

bond4u, 2017-02-12:
Machine zero marker would be nice.

fieldOfView, 2017-02-12:
2.4 has an origin indication, see https://github.com/Ultimaker/Cura/pull/1078#issuecomment-256182691
This indication moves to the center of the buildplate if machine_center_is_zero is checked

closed 2017-07-27 by nallath