Ultimaker/Cura #812: Cura 2.1.2 raft not showing up on multiple parts

Nasimovy, 2016-06-17:
i put 6 parts on the build plate, when slicing one of the parts does not have a raft
http://i.imgur.com/bugnZD5.jpg
all the parts are yellow
http://i.imgur.com/yAON5Xr.jpg
when i remove the part that has no raft another part gets no raft
http://i.imgur.com/xApemZ6.jpg
5 parts are still yellow
http://i.imgur.com/EnPzgMU.jpg

edit: after more testing it even happens with 2 parts

thopiekar, 2016-06-18:
1. Please give us a link with the parts or upload them somewhere if possible.
2. It seems that you are printing all parts after another. In that case please take care about the overlapping grayed sections: they shouldn't overlap!

Nasimovy, 2016-06-19:
http://www.thingiverse.com/thing:1559232  used the NT parts and 50% scaled.
About grey overlapping surfaces i thought this is the space needed by the printhead so it would not collide with an other printed part already on the build plate. So if the grey part is not touching the yellow part there is no problem

after even further testing with other STL's raft only shows up after several pieces on the build plate

= 1 part no raft
http://i.imgur.com/i7I3lgl.jpg
= 2 parts --> only 1 gets a raft
http://i.imgur.com/440qchb.jpg
= 3 parts --> only 2 get a raft
http://i.imgur.com/rKSSxXE.jpg

Nasimovy, 2016-06-19:
seems like a rendering bug, i tried to print the sliced thingies and guess watt a raft was printed. very strange bug.

Ghostkeeper, 2016-06-20:
Hmm, weird. Does it change when you zoom in, I wonder?

awhiemstra, 2016-06-20:
There is a known issue in 2.1 where the bottom most layers will not be visualized because the raft uses a negative layer number. This is fixed for 2.2. This looks to be the same bug. Since you are using one at a time mode, the other layers are created as layers > 0 and thus show up properly.

closed 2016-06-20 by awhiemstra