Ultimaker/Cura #814: Cura 2.1.2 - Won't slice this stl at 1.2mm shell

ZhuDaHai, 2016-06-20:
Got an object I made in Sketchup that will slice in Cura 15 (any shell thickness), but won't slice in Cura 2.1.2 when set to 1.2mm shell. It will slice at .8mm shell (nozzle size is .4mm) in Cura 2.1.2
I've vetted the object in Sketchup with Solid Inspector plugin with no errors.
Slicing stops at about 90% and then restarts, indefinitely.
Happy to supply .stl file (and skp file), if interested and its not an already known bug.
Thanks.

Ghostkeeper, 2016-06-20:
If it's 100% reproducible that is an interesting test case to help us debug, but the slice restarting is already known to us. It's a fickle one, I've heard.

ZhuDaHai, 2016-06-20:
Yes, 100% reproducible.
And another clue: If I set Top/Bottom Pattern to 'Concentric' it will slice with shell/wall thickness of 1.2mm. 'Lines' and 'Zig Zag' causes endless Slicing restarts.

Attached is a screenshot at the failure point. Also, the 'Close Window' X in the upper right turns red just before it restarts the Slicing.

Also attached is the stl file. Let me know if I can provide anything else.

Cheers.
[image]
 

[RPi 2 Case-v10.zip](https://github.com/Ultimaker/Cura/files/322915/RPi.2.Case-v10.zip)

ZhuDaHai, 2016-06-20:
Another clue: Z Seam Alignment set to anything but 'Shortest' also allows it to Slice at 1.2mm walls even when Top/Bottom Pattern is set to 'Lines'

awhiemstra, 2016-06-20:
The slicing loop in this case is probably happening because the engine crashes and the slicing process gets restarted. Not sure why the engine crashes though.

Ghostkeeper, 2016-06-20:
I've attached the model to the issue in our internal issue tracker (CURA-1652). Hopefully that helps people debug.

closed 2016-12-26 by thopiekar