Ultimaker/Cura #815: [master] wrong results in LayerView?

thopiekar, 2016-06-20:
I just noticed to much support in my LayerView, so enabled support only for "touching buildplate", and finally got this.

[image]

[image]

Did I miss something or is that a (known) bug?

nallath, 2016-06-20:
How does the model look in x-ray mode? It looks a bit like the model is faulty.

thopiekar, 2016-06-20:
[image]

thopiekar, 2016-06-20:
I prepared my model a bit and added some bevels. Now it looks fine :+1: 
[image]

closed 2016-06-20 by thopiekar