Ultimaker/Cura #832: Cura initial layer speed is applied to _all_ layers

gitjoost, 2016-06-30:
My Cura 2.1.2 initial layer speed is set to 20 mm/s and normal print speed to 50mm/s. However all layers are printed at the 20mm/s setting. Checked this by viewing Cura produced gcode in gcode viewer.  Needless to say; printing _anything_ is darn slow.

gitjoost, 2016-06-30:
A bit more detail: the setting I used is the standard, out of the box default setting for Normal Quality. Layers 3 and up are printed at 22.2mm/s - no idea where that comes from.

nallath, 2016-06-30:
Are there any "i" icons visible? It could be that you broke some inheritance stuff.

fieldOfView, 2016-06-30:
Are you by any chance printing at 100% infill? If so, try setting the infill at 99%

DDDirk, 2016-06-30:
Just to make sure, is it indeed slow for anything? Or just small objects? In that case it could be caused by the minimal layer time.

gitjoost, 2016-06-30:
@nallath: there was an "i" next to top/bottom speed stating that the parameter is normally computed but was set to an absolute speed. Clicking it made this go from 20mm/s to 22.2mm/s  --- this is the value I see with the gcode viewer. Funny thing is that this is the Normal Quality profile which is built-in and cannot be overwritten (supposedly).

@filedOfView: My infill parameters are:
Infill density: 20%
line distance 1.75mm
infill pattern: grid
infill overlap: 10%
infill wipe distance: 0.0875 mm
infill layer thickness 0.1mm
infill before walls: on

Speed parameters:
Print speed:50mm/s
infill speed:50mm/s
wall speed:40mm/s
outer wall speed: 30mm/s
inner wall speed: 40mm/s
top/bottom speed: 20mm/s (now 22.2mm/s after clicking the "i")
Travel speed: 120 mm/s
initial layer speed: 20 mm/s
skirt speed: 20 mm/s
Number of slower layers: 2

@DDDirk: The object I have is a flat back panel for an enclosure I am designing. It is 90mm square and 2.1mm thick. I added screenshot of the object as captured by Cura.
[image]

This particular object I have printed before in what I remember to be 1:30 or 2hrs with Cura 15. Now it prints in 5hrs with Cura 2.1.2.  I haven't printed much with Cura 2.1.2 yet and I have a  nagging thought that the first object I printed with this version (a different part of this enclosure assembly) did print at normal speed. 

I am fully capable of screwing things up so would love to be told where I did/what I did but looking the setup profile I cannot make sense of the gcode decisions Cura made and rather feel something went wrong there...

BagelOrb, 2017-07-23:
fixed a long time ago

closed 2017-07-23 by BagelOrb