Ultimaker/Cura #833: How to get rid of the lines around model ?

snuchitp, 2016-07-01:
[image]

I want to get rid of the three dark green lines around the model. What should I do?

Thanks

markwal, 2016-07-01:
@snuchitp I believe that is the skirt.  For version 15.04, you choose Platform Adhesion: None instead of Skirt.

fieldOfView, 2016-07-01:
For issues with Cura 15.04, please see https://github.com/daid/cura

closed 2016-07-01 by fieldOfView

snuchitp, 2016-07-01:
[image]

It is none for platform adhersion

snuchitp, 2016-07-01:
I click ... bottom at the end of platform adherson type. It pop up a new window as shown in a picture below that skirt line count =1

[image]

after I tried to change the value of skirt line count to zero. Those three lines are gone as shown in a picture below

[image]

Thanks